ArduPilot/ardupilot #21270: CRSF broken on 4.2.2 Matek F765 Wing

JacksonUAS, 2022-07-24:
## Bug report

**Issue details**

CRSF diversity nano outputting CRSF TX/RX to TX6/RX6 pins on F765.

Default param config
SERIAL7_PROTOCOL = -1 and 
BRD_ALT_CONFIG = 0

Have crossfire control, no telemetry.. which is curious as it shouldnt work.

Setting it up as per Wiki =

SERIAL7_PROTOCOL = 23 and / or BRD_ALT_CONFIG = 1 = No RC input.

Tried running CSRF to TX7/RX7 and changed Serial 1 protocol to 23 = No RC input.

Board/Firmware works on SPort etc. Wiring is correct. 

**Version**
4.2.2

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Vectored tailsitter

**Hardware type**
F765 Wing

xianglunkai, 2022-07-24:
I also noticed a similar problem！
It seems that serial7 could not longer work properly due to the recent code submission.
Maybe it's caused by the changed serial_manger class recently.

JacksonUAS, 2022-07-24:
Not sure if related, but CRSF VTX control is also not working (SERIAL4_PROTOCOL = 29).

Ive tried 4.3 Dev and 4.0 and neither are working.

xianglunkai, 2022-07-24:
There seems to be no connection, but I found that serial7 can't work normally under the curent master.

JacksonUAS, 2022-07-24:
SERIAL1 is dead as well, tried Mavlink and CRSF on it.

JacksonUAS, 2022-07-24:
SERIAL5 (UART8) is working for CRSF :)

There is something very wonky with F765s serial ports on Arduplane.

andyp1per, 2022-07-24:
You cannot use the CRSF VTX protocol and CRSF at the same time - if you want to have both you need to wire your VTX into the spare CRSF ports on your RX and then just use protocol 23 on AP.

Neither SERIAL1 nor SERIAL7 have DMA currently on this config so will not work with CRSF. You would need to use SERIAL2/UART1 as this is the only serial port with DMA.

Are you saying this worked in the past? If so what version worked?

JacksonUAS, 2022-07-24:
Ahh, none of that is documented :) The Arduplane wiki gives instructions to use Serial 7 for CRSF, but it definitely doesnt work.

SERIAL 5 definitely works with CRSF, and I just tried SERIAL 2 and it works as well.

mattdholmes00, 2023-03-18:
I can confirm, serial 7 worked for CRSF in 4.1.5. I just updated now to 4.3.5 with no other changes and CRSF telemetry stopped working.

andyp1per, 2023-03-18:
> I can confirm, serial 7 worked for CRSF in 4.1.5. I just updated now to 4.3.5 with no other changes and CRSF telemetry stopped working.

On this board?