ArduPilot/ardupilot #21271: uart communication is broken in current master branch

xianglunkai, 2022-07-24:
Serial communication work well before this submission!
[AP_SerialManager: remove mavlink-specific methods from AP_SerialManager]
 I repair AP_Serialmanager.cpp，AP_Serialmanager.h， GSC.h and GCS_ Common.cpp can restore the serial port to normal. 
Therefore, this submission  maybe needs further examination.
Txs!

andyp1per, 2022-07-24:
For reference it is this:

```
commit b958c3db70aec905f0d45134f513ca7153c0bc3b
Author: Peter Barker <pbarker@barker.dropbear.id.au>
Date:   Mon Jul 11 17:57:40 2022 +1000

    AP_SerialManager: remove mavlink-specific methods from AP_SerialManager
```

xianglunkai, 2022-07-24:
@andyp1per Thank you very much!

peterbarker, 2022-07-24:
Thanks for the report.

I'm unable to replicate any issues here.  For example, I flew master yesterday on both a quadcopter and Plane without issue, meaning basic mavlink and GPS communications are working.

AFAICS mavlink1 and mavlink2 work just fine in my testing here.

Could I get some more information on exactly what's not working for you, please @xianglunkai ?

xianglunkai, 2022-07-25:
@peterbarker  Yes, the current master branch of mavlink and GPS works well, but it does not work well for other types of 
 serial protocols, such as Rangerfinder serial.
Txs!

peterbarker, 2022-07-25:
> @peterbarker Yes, the current master branch of mavlink and GPS works well, but it does not work well for other types of serial protocols, such as Rangerfinder serial. Txs!

Hmmm..... I just plugged in a serially-connected rangefinder into a CubeOrange here and it worked as expected.  Getting readings without problems.  Still need some help replicating, it seems ;-)

xianglunkai, 2022-07-25:
@peterbarker I am sorry for rangefinder. Actually, this pr happens to the battery serial port by RS485.,
Further, I found that the problem is that defining uartstate as class rather than struct leads to this problem, but I am very confused about the difference between them.

xianglunkai, 2022-07-25:
I think this problem has been solved.
Thank you very much for your discussion. 
Please review the following request [https://github.com/ArduPilot/ardupilot/pull/21275]

peterbarker, 2022-07-25:
> I think this problem has been solved. Thank you very much for your discussion. Please review the following request [https://github.com/[/pull/21275](https://github.com/ArduPilot/ardupilot/pull/21275)]

Thanks very much for that patch!

To answer your earlier question, there are very few differences between classes and structs in C++ now-adays.  The big one is member visibility - members are public by default in a `struct`, private by default in a `class`.

peterbarker, 2022-07-25:
Closes https://github.com/ArduPilot/ardupilot/issues/21271

closed 2022-07-28 by xianglunkai