ArduPilot/ardupilot #21273: Automate reporting of firmware build limitations by board

Hwurzburg, 2022-07-24:
Maintaining the Wiki list of firmware limitations has become untenable and mistakes and ommisions are easily introduced...we need a script that will generate a list, by board type, of what features are excluded and listed in a file on the server which can be linked from the Wiki.....running this script every other day or less frequently will be sufficient....we need this ASAP

Hwurzburg, 2022-07-24:
@andyp1per suggests that the build server be enhanced to use that information to automatically select the configuration to match the current limitations list when a board is selected, making it easier to drop and add back features...

tridge, 2022-07-26:
https://ardupilot.org/copter/docs/common-limited-firmware.html

tridge, 2022-07-26:
i think processing the elf file at the time we build to generate a features.txt for each build is the first step

peterbarker, 2022-07-31:
https://github.com/ArduPilot/ardupilot/pull/21315