ArduPilot/ardupilot #21291: `handle_obstacle_distance_msg` ignores packet timestamp

obicons, 2022-07-27:
## Bug report

In [`handle_obstacle_distance_msg`](https://github.com/ArduPilot/ardupilot/blob/6ceee23b8537af71d57130015cb617ec005729d6/libraries/AP_Proximity/AP_Proximity_MAV.cpp#L118), the `time_usec` of the obstacle distance message is ignored. This is surprising behavior, since a malfunctioning subsystem  or transport delay could cause the autopilot to use stale data.

**Version**
Master

**Platform**
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
Any

**Logs**
N/A

magicrub, 2022-07-27:
An even worse scenario is a subsystem lying to us about the time... which is why we timestamp the data locally when the packet is received
https://github.com/ArduPilot/ardupilot/blob/6ceee23b8537af71d57130015cb617ec005729d6/libraries/AP_Proximity/AP_Proximity_MAV.cpp#L141

Closing for now, we can re-open if you think this deserves more discussion.

closed 2022-07-27 by magicrub

obicons, 2022-07-27:
>An even worse scenario is a subsystem lying to us about the time

Do we really need to be concerned about this? If the subsystem is able to send this message, then it can send arbitrary MAVLink messages. The common case is delay, so I think it makes sense to trust subsystems.

magicrub, 2022-07-27:
pull requests welcome!