ArduPilot/ardupilot #21303: LOITER_UNLIM next command

pavloblindnology, 2022-07-28:
## Bug report

**Issue details**
I'm having troubles with the next mission
1. TAKEOFF
2. LOITER_UNLIM
3. WAYPOINT

Once copter reaches LOITER_UNLIM waypoint I cannot proceed to the next waypoint by toggling the flight mode between Auto and Stabilize/AltHold/Loiter/Guided, as it should according to [manual](https://ardupilot.org/copter/docs/common-mavlink-mission-command-messages-mav_cmd.html#mav-cmd-nav-loiter-unlim):
>The mission will not proceed past this command while in AUTO mode. In order to break out of this command you need to change the mode (i.e. to MANUAL). If there are subsequent commands then you can continue the mission at the next command, if the Copter MIS_RESTART parameter is set to resume, by switching back to AUTO mode (otherwise the mission will restart).

The issue was discussed with no answer: https://discuss.ardupilot.org/t/loiter-unlimited-next-command/8719/2

**Version**
Copter_4.2.2

**Platform**
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
octa

**Hardware type**
SITL

IamPete1, 2022-07-28:
I suspect this is a case were we just need to update the documentation. What behavior are you trying to achieve?

Using scripting you could add a RC switch to advance past a loiter unlimited. If you have a GCS you can just advance the mission index directly.

magicrub, 2022-07-28:
The current behavior is correct, it's the doc that is wrong. Switching out of AUTO and back will not skip over a mission waypoint. It will either resume where it left off or reset to beginning depending on the `MIS_RESTART` param.

pavloblindnology, 2022-07-29:
> I suspect this is a case were we just need to update the documentation. What behavior are you trying to achieve?
> 
> Using scripting you could add a RC switch to advance past a loiter unlimited. If you have a GCS you can just advance the mission index directly.

Understood. The idea is to do some manual/guided piloting at some of the waypoints, then continue with the mission. Switching back to Auto is the simplest way, not requiring any scripting or GSC.
Is there any reason behind not proceeding to the next waypoint after LOITER_UNLIM? I understand, that 2 cases are possible here: wp is reached - then we should not proceed to the next wp; wp is not reached - then we should proceed.

IamPete1, 2022-07-29:
If you fancy doing a PR that would be great, no guarantees we will accept it tho. 

You could do it based on the mode change from scripting too.

magicrub, 2022-07-29:
@pavloblindnology How about a new AUX switch input to skip to next waypoint? I thought we already had that but I don't see it listed on the wiki. That is a fairly simple thing to implement. Would that work for your application?
https://ardupilot.org/copter/docs/common-auxiliary-functions.html

Another option, as @IamPete1 mentioned, is a simple LUA script that you write and drop onto the SD card to do the equivalent of reading an RC input switch to move the waypoint index forward or back. Lots of examples here showing RC in and changing the mission.
https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_Scripting/examples

pavloblindnology, 2022-07-31:
@magicrub @IamPete1 Sure, could do this with scripting. Just thought it'd be more convenient for an operator to proceed with 1 switch instead of 2.

IamPete1, 2022-07-31:
No reason you can do it based on mode change in the script as well.