ArduPilot/ardupilot #21310: Reversed sticks after software update

shaikon, 2022-07-29:
## Bug report

**Issue details**

https://discuss.ardupilot.org/t/reversed-yaw-roll-and-pitch/87649

**Version**
4.2.1

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X Quad

**Hardware type**
Pixhawk

IamPete1, 2022-07-29:
@shaikon Can you upload a param set from before the update and one from after. What RC protocol are your using?

shaikon, 2022-07-30:
I'm using a DX6i Radio - so I guess it's DSM.
How can I download the param set values into a file so I can upload them here ?

IamPete1, 2022-07-30:
If you have a dataflash log from the SD card from before and after we can extract the params. 

https://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html#downloading-logs-via-mavlink

shaikon, 2022-07-30:
Unfortunately I don't have the BEFORE file. But I can of course download the current. Would that help you ?

IamPete1, 2022-07-30:
We really need both before and after to see what when wrong in the update. If your feeling especially keen you could flash back to 4.1, get a log, and try the update again.  I presume it was 4.1 you updated from?

shaikon, 2022-07-30:
No. It was a very old version : 3.3.3
Should I try to revert back to it as you suggested ?

IamPete1, 2022-07-30:
Ah, that is certainly the issue then. We don't try and do the correct thing over such a big update (5 years since 3.3). It will  work from one stable to the next, but no guarantees over multiple versions. 

If you did have to update you need to go via the releases in between. IE 3.3 -> 3.4 -> 3.5 -> 3.6 -> 4.0 -> 4.1 -> 4.2.

But there have been so many improvements you would be better just doing a fresh install repeat the initial setup. You should be able to get significantly better results.

closed 2022-07-30 by IamPete1

shaikon, 2022-07-30:
Sure. I understand that. But I'm still curious why the sticks are reversed when doing the update....
As far as I understand when installing v4.2.1 on a fresh FC or upgrading from something very old and doing the initial setup according to the requirements - one should NOT end up with reversed pitch/yaw/roll sticks.

I mean, assuming proper setup and calibration - a ROLL LEFT on the radio should ALWAYS cause the vehicle to ROLL LEFT.

geofrancis, 2022-07-30:
@shaikon the problem is no one else is seeing this issue, I suspect you have something reversed somewhere that has been unreversed by the update.

shaikon, 2022-07-30:
"Unreversed by the update" - interesting idea. I'll gladly keep looking what's wrong on my end...
However, I already double checked that the RC isn't reversed in it's LCD menu and I also verified that the reversal options in mission planner aren't turned on.

Regarding "no one else is seeing this issue".
When I posted on the community forum someone even called this problem "common":
https://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html#downloading-logs-via-mavlink

IamPete1, 2022-07-30:
Its common that peoples RC channels are backwards, the wiki has instructions on how to correct this. 

https://ardupilot.org/copter/docs/common-radio-control-calibration.html#radio-control-calibration

Its not ideal that something was changed when you updated but were not going to put any effort in to fixing issues that happen when you update to 4.2 from 3.3. 

If the same thing happens on the 4.1 to 4.2 update then we will certainly look into and fix it.

geofrancis, 2022-07-30:
yes its common to need the pitch reversed on most radios, its not common to have things get reversed when updating.

shaikon, 2022-08-01:
> ardupilot is almost totally new software since version 3 I'm surprised you got as far as it taking off using parameters as old as that.

I went through initial calibration afterwards.