ArduPilot/ardupilot #21311: Implement response to mavlink ping (requests)

Consti10, 2022-07-29:
## Feature request

Respond to MAVLINK_MSG_ID_PING (requests). This allows for a simple way to measure latency from any mavlink system to the FC.

I am not sure, but it seems as if my FC running ardupilot doesn't respond to mavlink ping messages.
https://mavlink.io/en/services/ping.html

Adding support for that would be really easy, see here for a simple example implementation:
https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/ping.cpp#L41

I've tried searching the ardupilot code for both MAVLINK_MSG_ID_PING and mavlink_msg_ping_decode, without any results.
Or am I missing something ?

**Platform**
[ x ] All

WickedShell, 2022-07-29:
> This allows for a simple way to measure latency from any mavlink system to the FC.

You can use `TIMESYNC` to measure the latency as well, which is supported, and should actually give you a better estimation of timing errors.

Consti10, 2022-07-29:
True, but timesync is much more complicated (e.g on the groundstation side) and also takes up bandwidth - aka overkill if your goal is just to ping the FC once and get a round-trip latency value.
 
Handling this kind of message is really simple - adding it should only be a (slightly modified) copy - paste of the code I already linked. Not sure if that's worth a pull request.

peterbarker, 2022-07-30:
> True, but timesync is much more complicated (e.g on the groundstation side) and also takes up bandwidth - aka overkill if your goal is just to ping the FC once and get a round-trip latency value.

As @WickedShell says - `TIMESYNC` is what you should be using.

`PING` is deprecated in the XML and will be removed from the standard at some stage in the future.

> Handling this kind of message is really simple - adding it should only be a (slightly modified) copy - paste of the code I already linked. Not sure if that's worth a pull request.

Yes, you're correct, it would be straight-forward to add.  But it will cost us in flash bytes which aren't necessary for us to carry and *every board* has to carry the burden of those bytes.  Some of our boards this can really matter!

`timesync` isn't *that* complicated to implement on the client side :-)  You don't even need to fill in the tc field with anything that matters IIRC.  Here's MAVProxy's implementation: https://github.com/ardupilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_system_time.py#L76

Incidentally, we do have a low-volume google group where we announce changes to our mavlink protocol handling; you might consider joining it.

Consti10, 2022-07-30:
Thanks for clarifying that. 
After looking a bit at the timesync protocoll, my first thought was that it would be pretty straight forward to calculate the round trip latency by using a single timesync message - aka. use timesync for a simple "ping".

I am quite sure that should be possible - since the mavlink timesync request (tc1==0) and response (original timestamp + tc1!=0) is basically the same underlying idea as with ping.

However, r.n I am only receiving timesync messages in my ground station with (tc1==0) aka no responses to my original timesync message. 

Is this the ardupilot code responsible for timesync:
https://github.com/ArduPilot/ardupilot/blob/plane3.8/libraries/GCS_MAVLink/GCS_Common.cpp#L1723

?

Consti10, 2022-07-30:
> Thanks for clarifying that. After looking a bit at the timesync protocoll, my first thought was that it would be pretty straight forward to calculate the round trip latency by using a single timesync message - aka. use timesync for a simple "ping".
> 
> I am quite sure that should be possible - since the mavlink timesync request (tc1==0) and response (original timestamp + tc1!=0) is basically the same underlying idea as with ping.
> 
> However, r.n I am only receiving timesync messages in my ground station with (tc1==0) aka no responses to my original timesync message.
> 
> Is this the ardupilot code responsible for timesync: https://github.com/ArduPilot/ardupilot/blob/plane3.8/libraries/GCS_MAVLink/GCS_Common.cpp#L1723
> 
> When I read the code, it should work:
> 
> 1. GroundControll sends timesync message with tc1==0 and ts1==currGroundStationTimeNanos()
> 2. FC gets this message and responds with tc1==currFcTimeNanos and ts1==original.ts1
> 3. GroundControll gets message with ts1==last outgoing timestamp and therefore knows this is a response to it's request and can
>    calculate the round trip latency.
> 
> Only for some reason, it doesn't. Are there any other requirements on the FC to respond to timesync, e.g. must it be armed or similar ?
> 
> Aka r.n I am only getting timesync "requests" from ardupilot, no responses. But I'l keep debugging, could be an issue on my end, too.

... and of course, was my programming error. (sent the wrong type of message). Now it works flawlessly.

Aka the short conclusion here is:
1) Ardupilot does not support PING
2) But it supports timesync
3) with timesync, you can easily perform a ping operation, making 1) obsolete.

Right ? I think the issue can be closed.

peterbarker, 2022-07-30:
>     1. Ardupilot does not support PING
>     2. But it supports timesync
>     3. with timesync, you can easily perform a ping operation, making 1) obsolete.
> 
> 
> Right ? I think the issue can be closed.

Yep, all good, thanks for the thorough follow-up.

n.b. One thing to note with timesync - there's no target field in the message.  That means you can't tell whether a message you receive with a filled-in ts1 field actually corresponds to the message you sent.  If you are using millisecond-resolution timestamps in the sent message, that does mean if there's another mavlink out there also sending requests (and there are at least three nodes on the segment) @1Hz there's a ~ 1/1000 chance you'll get see someone else's response as your own.  That's what the "+1234" thing in the mavproxy module was all about.... playing with the concept of using the low-order bits to tag the sender's ID into the echoed field.

You probably don't care about all that if you're just doing a ping-for-aliveness thing :-)

Just found Hamish is putting a document together here: https://github.com/mavlink/mavlink-devguide/pull/433/files

Closing this issue now.

closed 2022-07-30 by peterbarker

Consti10, 2022-07-31:
> Yep, all good, thanks for the thorough follow-up.
>
> n.b. One thing to note with timesync - there's no target field in the message. That means you can't tell whether a message you receive with a [...]

Yeah, you are totally right. I'd say this is a general problem of the (timesync) protocoll definition though. Aka, as you said, highly unlikely (especially when going for microseconds / nanoseconds) but still, possible. Ping is more resilient in this regard. Interesting that's the "workaround" in mavproxy, I totally would have missed that if you didn't point it out.

I'd vote for changing that in the "timesync" protocoll specification - but I don't know if the couple of bytes needed (either a sequence number that is also packed into the response, or similar) are worth it.