ArduPilot/ardupilot #21319: Plane: auto takeoff timeout does not timeout

teodoryantcheff, 2022-07-31:
## Bug report

**Issue details**
Timeout value for takeoff is discarded. Motor spins indefinitely.

To reproduce:

1. remove prop
2. set Takeoff as mode (77) on arbitrary rc channel.
3. set-up shake to wake.
4. set TKOFF_TIMEOUT. I used 5 seconds.
5. arm plane.
6. "activate" takeoff mode channel.
7. shake to wake. The motor spins as expected.
8. leave the plane on the ground.
9. wait for more than the set TKOFF_TIMEOUT.
10. Takeoff mode is not cancelled, Motor spins indefinitely, plane remains armed.

**Version**
plane 4.2.1

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Matek f765wing

teodoryantcheff, 2023-11-12:
Just now tested the same with Plane 4.4.2. `TKOFF_TIMEOUT` is still not taken into account and the motor spins indefinitely with the model on the ground.
No airspeed sensor, but a solid ~10 sats GPS lock.

magicrub, 2023-11-13:
I'll take a look

teodoryantcheff, 2024-06-19:
> I'll take a look

Did you, @magicrub ?