ArduPilot/ardupilot #21321: Crash with Baro(MS5611) update function running Ardupilot on BBB

juvinski, 2022-07-31:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

After updating from 4.1 to dev(4.3.3) - running arducopter - the arducopter is freezes - with no lights blinking, any GSC response/message.
Trying running the examples one by one, I got on the BARO a segfault 

In this same board with same linux and  same kernek  I have the 4.1 final and is working fine with any crash or something.

**Version**
Dev version

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[ X ] Plane
[ ] Rover
[  ] Submarine

**Airframe type**
Quad X

**Hardware type**
Beaglebone Back and Green

**Logs**

Starting Ardupilot I'm having:

Config Error: Baro: unable to initialise driver

And running the example - BARO_generic

Starting program: /root/ardupilot/build/bbbmini/examples/BARO_generic 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

[New Thread 0xb6cf4450 (LWP 1583)]
[New Thread 0xb69ff450 (LWP 1584)]
[New Thread 0xb66ff450 (LWP 1585)]
[New Thread 0xb65fe450 (LWP 1586)]
Barometer library test
MS5611 found on bus 2 address 0x01
[New Thread 0xb6bf3450 (LWP 1590)]
TOGCS: Calibrating barometer

Thread 1 "BARO_generic" received signal SIGSEGV, Segmentation fault.
0x0040d836 in AP_Logger::should_log(unsigned int) const ()
(gdb) backtrace
#0  0x0040d836 in AP_Logger::should_log(unsigned int) const ()
#1  0x00408bc4 in AP_Baro::update() ()
#2  0x00485dc8 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 

Any clue ?

juvinski, 2022-08-07:
I'm facing several problems with SPI and beaglebone.
First was this crash, now the crash continues in the examples, but ardupilot is running on the master/dev branch.
But the load average is hudge - and all examples are running fine and with almost none load average, but anything running with SPI bus the load average increase exponentially.

palladin82, 2022-08-08:
Try to change DMA_PRIORITY in [ardupilot](https://github.com/ArduPilot/ardupilot)/[libraries](https://github.com/ArduPilot/ardupilot/tree/master/libraries)/[AP_HAL_ChibiOS](https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_HAL_ChibiOS)/hwdef/YOUR_HARDWARE/hwdef.dat

like this DMA_PRIORITY SDMMC* USART2* ADC* SPI* TIM*
to this DMA_PRIORITY SDMMC* SPI* USART2* ADC*  TIM*

juvinski, 2023-08-28:
There problem was fixex by the PR https://github.com/ArduPilot/ardupilot/pull/24786

closed 2023-08-28 by juvinski

qm1101uetclone, 2025-03-29:
Hi, I am having problem with ms5611
debian@beaglebone:~$ sudo /home/debian/BARO_generic
Barometer library test
Config Error: Baro: unable to initialise driver
TOGCS: Config Error: Baro: unable to initialise driver
I hope you can help me. Thanks bro

juvinski, 2025-04-02:
Hi @qm1101uetclone 
are the /dev/spidev ok ? what devices do you have ?

qm1101uetclone, 2025-04-03:
I checked, below is the information I checked. Pls help me. Thanks

> Message ID: ***@***.***>
>

qm1101uetclone, 2025-04-03:
you can check it. I hope you can help me. Thanks. 
https://files.catbox.moe/cmdz14.png
https://files.catbox.moe/hpnf4n.png