ArduPilot/ardupilot #21336: #21282 could prevent flying plane and quadplanes without compasses from ever flying

Hwurzburg, 2022-08-03:
I think this will prevent me flying most of my planes without compass and one quadplane....on the ground without compass, until EKF3 aligns yaw in the initial flight or hover, it stays in DCM....so you cant arm in compass-less planes....I think....havent flown this code but just going by how I think things work....worth considering....maybe needs a compassless (no compass uses) to bypass this check?

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

tridge, 2022-08-03:
you're right, needs fixing

Hwurzburg, 2022-08-03:
@tridge tested on real aircraft with master....compass use =0.....sometimes on the bench EKF3 will be shown as active INS after boot and it arms...havent moved it like walking to flight line but as I recall it can revert to DCM while doing that or waiting on the ground..., but almost always if you disarm after that it will go to DCM and present the prearm....sometimes it will toggle back and forth between DCM and EKF while sitting disarmed..if in DCM state, you cant arm....this is with good gps and all other prearms absent....so it can prevent arming on a compass-less setup

Hwurzburg, 2022-08-03:
Confirmed outside....often you can arm after first powerup and it will be using EKF3...but definitely after the disarm, it will toggle back and forth EKF3 and DCM with good sats and HDOP, and in DCM periods you cant rearm....I think it happens after first power up also, ie toggling, but in my one  test outside (its 40C! so did not spend a lot of time)  it did not toggle back and forth...but I recall seeing that on the bench at the field after first power up on compassless planes....

Hwurzburg, 2022-08-29:
fixed...closing

closed 2022-08-29 by Hwurzburg