ArduPilot/ardupilot #21382: lua rc:override binding doesn't work in real cube, but works in sitl

davidbuzz, 2022-08-08:
## Bug report

When loading /libraries/AP_Scripting/examples/RC_override.lua into ardupilot on a Cube, it crashes.

_Please describe the problem_

**Version**
ArduCopter V4.2.2 stable(4fcfa4b2) -  but it probably doesn't matter ??

**Platform**
[  X] All
[  ] AntennaTracker
[  X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Cube Orange, but it doesn't matter so long as sescripting is enabled.

**Logs**
8/08/2022 5:40:39 PM : RC_override example
8/08/2022 5:40:39 PM : Lua: /APM/scripts/RC_override.lua:8: attempt to call a nil value (method 'set_override')

IamPete1, 2022-08-08:
It only recently went in, https://github.com/ArduPilot/ardupilot/pull/20241 Looks like its just not in stable?

davidbuzz, 2022-08-08:
It went in master in March, Copter stable was released in May, should b there

IamPete1, 2022-08-08:
It was close, 4.2 went into beta at the start of March. The bindings were merged at the end of March.

davidbuzz, 2022-08-08:
:-(

closed 2022-08-08 by davidbuzz