ArduPilot/ardupilot #21389: ArduRover for Pixhawk 6C

jakobjackson2016, 2022-08-09:
## Feature request

**Is your feature request related to a problem? Please describe.**
_I have been seeing a `latest` ArduRover release for HolyBro Pixhawk 6C for a long time but it has been failing `test.Rover` Autotest: https://autotest.ardupilot.org/_

**Describe the solution you'd like**
_Is there an ETA for a stable ArduRover release for HolyBro Pixhawk 6C? I am waiting for a stable release for Pixhawk 6C to build a rover_

**Describe alternatives you've considered**
_In the past, I have build ArduPilot based rover with Pixhawk 4 and it worked flawlessly. Unfortunately, Pixhawk 4 production has been out of stock for a long time and only HolyBro board available is Pixhawk 6C.  I have been waiting for two months for ArduRover for Pixhawk 6C to release._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

**Additional context**
_ Also, has anyone deployed ArduRover on Pixhawk 6C successfully? I am afraid to deploy it to replace Pixhawk 4 board since the `latest` ArduRover release is not `stable`._

rmackay9, 2022-08-09:
Hi @jakobjackson2016,

Support for the Pixhawk6 will be included in 4.2.3 which will start beta testing by early next week at the latest.  Until then you can load "latest" onto the autopilot if you'd like to test it.

jakobjackson2016, 2022-09-26:
Thanks! I see that ArduRover 4.2.3 for Pixhawk 6C has been released. I will close this issue now.

closed 2022-09-26 by jakobjackson2016