ArduPilot/ardupilot #21408: Cruise Mode for Copters

devanesanandrews, 2022-08-11:
## Feature request
     Cruisie Mode for Copters like it is available in planes

I have been using Copters and planes  for fpv flying, Though autonomous mision and Guided modes are availbe in Copter, Cruise mode in planes are always easy to do long range

Adding Cruise mode, to copter would be good for long range flying as it does it in plane.

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2022-08-11:
@devanesanandrews,

Txs for the enhancement request.  What's the advantage of Cruise over an existing mode like Loiter?  Is it that the user doesn't need to keep the pitch stick forward?

devanesanandrews, 2022-08-12:
> @devanesanandrews,
> 
> Txs for the enhancement request. What's the advantage of Cruise over an existing mode like Loiter? Is it that the user doesn't need to keep the pitch stick pushed forward?

Thank you rmackay9 for looking at this,

Yes its is one of the reason, i have been flying with both fixed wing and copter, Cruise mode in Fixed was pretty easy and there is no need for constant input even in throttle,(Mostly i fly with self centering throttle).

Copter we have to constantly keep the throttle pushed.

Thanks