ArduPilot/ardupilot #21415: Copter: pre-arm check that throttle is high should be skipped if arming in Auto

rmackay9, 2022-08-12:
AP_Arming_Copter::arm_checks() skips over the pilot's throttle position checks if arming in Guided mode from a GCS.  I think we can also skip these checks if the vehicle is arming in Auto mode.  [See code here](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/AP_Arming.cpp#L641).

AltonNewbie, 2022-08-12:
Can ardu be used in conjunction with NVIDIA JETSON NANO for a lawn mower installation so it will (Recognize obstacles with cams/sensors on nvidia side and ai memorize mowing around object?
Though I first load mowing parameters (object stop it) I feel nvidia takes over to go around object and back in line with ardu parameters to continue mowing.
Plus I would code in the nvidia to identify all objects for safety.  If dog or human approaches, power off mower “UNTIL” HUMAN OR ANIMAL IS ATLEAST 40 FEET AWAY FROM MOWER.
PLEASE ADVISE..
ALTON

Smart AssistAlton D Davis Jr483 LAMKIN RDHARLEM GA 30814.     (770) 651-7009. [10am -10pm Eastern]

agising, 2022-09-08:
Thumbs up, I want this feature. Actually in my checklist I have 'throttel to mid-stick" prior to gcs takeoff

magicrub, 2023-01-17:
looks like @prathamesh7255 has a fix for this in PR https://github.com/ArduPilot/ardupilot/pull/22676

khancyr, 2023-01-18:
We got to PR that are the same and work, but it need some param change : 
AUTO_OPTIONS 1 to allow arming
RC_OPTIONS 32 to disable throttle arming check on RC lib.

rmackay9, 2023-01-18:
Opps, @khancyr you think the earlier PR isn't sufficient?

closed 2023-01-18 by rmackay9

khancyr, 2023-01-18:
It is needed but that isn't enough on its own as the rc_options check will trigger  but that is just param so the PR was good

PR #21994 (2023-01-18): Copter: skip throttle high arm check when arming in AUTO
Fixes #21415 
Hi @rmackay9! I am a newbie and have tried to add the necessary conditions. Please let me know if this is what you were looking for.
Thanks in Advance.