ArduPilot/ardupilot #21417: RC Option 5 (ELRS ARM/DISARM)

adalta87, 2022-08-12:
## Feature request

**Is your feature request related to a problem? Please describe.**
Not necessarily related to a problem but could cause issues given ELRS need for ch5 to be high for many things to function properly.

**Describe the solution you'd like**
Due to the necessity of ELRS needing ch5 to be high I think it would be beneficial to have an rc option for ch5 like we do for 6,7,8,9, etc..

**Describe alternatives you've considered**
I have just made a radio mix to keep ch5 high but seems messy and would be nice to have rc 5 option for arm/disarm.

**Platform**
[X] All
[  ] AntennaTracker
[ ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
⚠ Put your arm switch on AUX1 ⚠. AUX1 is the low-latency switch, sent with every packet, and only supports on/off (2-position) operation. ExpressLRS uses AUX1 to determine if your model is armed, and should set up on the transmitter so that ~1000 is disarmed, ~2000 is armed.
https://www.expresslrs.org/2.0/software/switch-config/
https://www.expresslrs.org/3.0/software/switch-config/

IamPete1, 2022-08-12:
There are RC options for all RC inputs.

closed 2022-08-12 by IamPete1

adalta87, 2022-08-12:
Are they just set in the parameters tree instead from extended tuning?

IamPete1, 2022-08-12:
Right, you will have to use full list or tree. RC5 is typically the flight mode channel so was never added to the "quick" access pages.