ArduPilot/ardupilot #21438: Barometer TCAL not working

harkhka, 2022-08-14:
## Bug report

I was trying to do baro temperature calibration and according to the wiki ([https://ardupilot.org/copter/docs/common-baro-temp-comp.html], I set the TCAL_ENABLED to 2 and powered down the drone to let it cool, and then powered it up and kept for 12 minutes without moving.

Then when I see TCAL parameters, they are not updated at all! I tied the complete process 2 times but still the same issue.

Using Arducopter 4.2.2 on MatekH743 board

**Version**
_What version was the issue encountered with_

**Platform**
Copter 4.2.2

**Airframe type**
V Quad

**Hardware type**
Matek H743 Mini

IamPete1, 2022-08-14:
You have to cover at least 10 deg. The instructions were written with boards with heaters in mind, in which will cover a much wider temp range in room temperature ambient conditions. Since there is no heater on a MatekH743 you will have to "manualy" cover the temp range. Freezer to room temp should do it. Enable disarmed logging and we will be able to tell how it got on from the log.

harkhka, 2022-08-14:
> 

Hello,

Yes I'm aware about the temperature range.
I kept it in my fridge for around 5 minutes and then put it inside a closed cardboard box for approx 12 minutes.
This whole time, the board was powered by a 4S battery, and because of the voltage regulators it reached around 65-70 degrees Celsius. It was very hot to touch for more than 2 seconds.

So I think that I was good on the temperature range part.

harkhka, 2022-08-21:
@rmackay9

hyperion11, 2023-05-13:
[image]
Flywoo GOKU GN 745 AIO custom firmware with tcal enabled. IMU working, baro not.

GalacticGreg, 2024-01-22:
I can confirm that TCAL is not working. I used the same method I used for IMU Temperature calibration (1. put it in the freezer for an hour or two 2. take it out and plug it in 3. leave it still for an hour to warm up.) No messages and no updates for the calibration parameters.

andyp1per, 2024-01-23:
TCAL for baro is not designed to work this way. So currently its a feature not a bug. It should ideally work the same way as IMUs but that's a bunch of work.

GalacticGreg, 2024-01-23:
I should add that I had the board plugged into the usb cable so I could plug it into the computer without moving it after the hour. I did this just in case I had to manually switch tcal_enabled to 1 once I saw a calibration value reported, no value was reported though.

Andyp1per, I followed the wiki pages instruction to do this. Did I misinterpret something or does the page need updating?

I know the page also says that this is not necessary, and based on how my quad flies, it is clearly not, but calibrating is satisfying ;)

amilcarlucas, 2024-01-31:
5 minutes in the freezer is too little. Do it overnight for 8 hours and place it in the kitchen oven, (without ventilation) at 50 °C.
Wait for 15 Minutes. and tell us the results.

GalacticGreg, 2024-01-31:
The IMU temperature was sensing -18 Celsius after being in the freezer for a couple of hours, so that should be sufficiently cold for the barometer as well. I let it warm up until the IMU read 45 Celsius. Is the oven required to heat the board faster or to get it to a higher temperature?

I don't want to do this step on a "let's just see what happens" approach. Unless someone familiar with the programming tells me that using a oven is what the code is designed to do.