ArduPilot/ardupilot #21449: create_terrain.py forever looping at "waiting on download of 41,-74"

Rand0409, 2022-08-15:
## Bug report

**Issue details**

Using create_terrain.py with "python .\create_terrain.py --lat=40.776676 --lon=-73.971321 --radius=1 --spacing=30", after several minutes downloading, the .DAT file size reached 31724KB, and the (lat_e7, lon_e7) in function create_degree reached to (41, -74). The script stopped downloading, and started looping "waiting on downloading of 41,-41". Should it be stopped automatically or forever loop?  

**Version**
latest version code from github

**Platform**
Windows 11, python3

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
no hradware used yet, only on windows

**Logs**
e7 set 409960577 -739996430
e7 set 409960577 -739992859
e7 set 409960577 -739989289
e7 set 409960577 -739985718
e7 set 409960577 -739982147
e7 set 409960577 -739978577
e7 set 409960577 -739975006
e7 set 409960577 -739971435
e7 set 409960577 -739967865
e7 set 409960577 -739964294
e7 set 409960577 -739960723
e7 set 409960577 -739957153
e7 set 409960577 -739953582
e7 set 409960577 -739950011
e7 set 409960577 -739946441
e7 set 409960577 -739942870
e7 set 409960577 -739939300
e7 set 409960577 -739935729
e7 set 409960577 -739932158
e7 set 409960577 -739928588
e7 set 409960577 -739925017
e7 set 409960577 -739921446
e7 set 409960577 -739917876
e7 set 409960577 -739914305
e7 set 409960577 -739910734
e7 set 409960577 -739907164
e7 set 409960577 -739903593
e7 set 409960577 -739900022
e7 set 409960577 -739896452
e7 set 409960577 -739892881
e7 set 409960577 -739889310
e7 set 409963271 -740000000
e7 set 409963271 -739996430
e7 set 409963271 -739992859
e7 set 409963271 -739989289
e7 set 409963271 -739985718
e7 set 409963271 -739982147
e7 set 409963271 -739978577
e7 set 409963271 -739975006
e7 set 409963271 -739971435
e7 set 409963271 -739967865
e7 set 409963271 -739964294
e7 set 409963271 -739960723
e7 set 409963271 -739957153
e7 set 409963271 -739953582
e7 set 409963271 -739950011
e7 set 409963271 -739946441
e7 set 409963271 -739942870
e7 set 409963271 -739939299
e7 set 409963271 -739935729
e7 set 409963271 -739932158
[... 419 lines cut ...]
e7 set 409998306 -739917873
e7 set 409998306 -739914303
e7 set 409998306 -739910732
e7 set 409998306 -739907161
e7 set 409998306 -739903590
e7 set 409998306 -739900020
e7 set 409998306 -739896449
e7 set 409998306 -739892878
e7 set 409998306 -739889307
e7 set 410001001 -740000000
0
waiting on download of 41,-74
0
waiting on download of 41,-74
0
waiting on download of 41,-74
0
waiting on download of 41,-74
0
waiting on download of 41,-74
0
waiting on download of 41,-74
0
waiting on download of 41,-74
0

stephendade, 2022-08-25:
You may need to install the latest version of MAVProxy (1.8.54), as some of the terrain support libraries have been changed recently.

Rand0409, 2022-08-25:
> 

I will try that, thank you!

Rand0409, 2022-08-25:
> 

Hi I tried with 1.8.54 version MAVProxy, still same issue happens

stephendade, 2022-09-01:
Ahh, just realised that you're using Windows. I've now confirmed that ``create_terrain.py`` doesn't work on Windows.

We do have an online terrain server at https://terrain.ardupilot.org/

Rand0409, 2022-09-06:
Hi,  I will try on Ubuntu, thank you very much!

Rand0409, 2022-09-06:
It is now working properly on Ubantu, python2, MAVProxy 55 version

closed 2022-09-08 by Rand0409