ArduPilot/ardupilot #21468: The UART3 serial port in CUAV Nora is not functional

maciek252, 2022-08-17:
## Bug report

The UART3 serial port (the one on the longer side) in CUAV Nora is not functional. The discussion is here:
https://discuss.ardupilot.org/t/nora-serial-port-numbering/73906/9
(it was mentioned in the final post that this issue will be fixed in the new firmware, but it has not been so far)

**Version**
4.2.0 and later

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

maciek252, 2022-08-17:
After submitting I've found this one:

https://github.com/ArduPilot/ardupilot/pull/20395

but I can't see this fix mentioned in the release notes, and the port is still not functional in 4.2.2

peterbarker, 2024-11-24:
No answer from original author, and I'd be astonished if this port wasn't functional.

Closing this one now - feel free to re-open if the port doesn't work on 4.5.7 (or the 4.6 beta)

closed 2024-11-24 by peterbarker