ArduPilot/ardupilot #21469: Acceleration position targets are not responded to in a timely fashion

ardugitter, 2022-08-17:
## Bug report

**Issue details**

If you set an acceleration position target (in this case I have been testing in SITL with MAVLINK_MSG_ID_SET_POSITION_TARGET_GLOBAL_INT), the drone will accelerate in the appropriate direction however when you reverse the polarity of the acceleration target, the drone will take much longer to respond and with a weaker acceleration and the end result is that the drone keeps flying in the same direction, slowing down only slightly (and not coming to a halt as you would expect).

In this case I did this (pseudo code):

```
main()
{
takeoff()
set_yaw()
sleep(4)

for(int i=0; i < 2; i++)
{
set_accel(5.0, 0, 0);
msleep(1000);
set_accel(5.0, 0, 0);
msleep(1000);
set_accel(-5.0, 0, 0);
msleep(1000);
set_accel(-5.0, 0, 0);
msleep(1000);
}

land()
}

set_accel(x, y, z)
{
type_mask = POSITION_TARGET_TYPEMASK_VELOCITY_IGNORE | \
                    POSITION_TARGET_TYPEMASK_POSITION_IGNORE | \
                    POSITION_TARGET_TYPEMASK_YAW_IGNORE | \
                    POSITION_TARGET_TYPEMASK_YAW_RATE_IGNORE;
afx=x;
afy=y;
afz=z;
}
```

Yaw is included in the position target to always maintain a 0 degree bearing. WPNAV_JERK=20 in case it matters (I didn't see any effect of increasing this from the default of 1 to 20).

The same behaviour is seen when including a velocity target of the same size and sign as the acceleration in the same Mavlink message.

**Version**
4.3.0dev SITL but also 4.2.2 on a real drone

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
SITL

**Logs**
This is a log from the SITL test:

[2022-08-17 13-11-35.bin.zip](https://github.com/ArduPilot/ardupilot/files/9359424/2022-08-17.13-11-35.bin.zip)

If you look at the the east velocity, you will see:
1) An acceleration for **2.9** seconds 
2) An acceleration in the reverse direction for 1.9 seconds (as expected) BUT at much less magnitude.
3) Same as 2) but reversed direction.
4) Same as 2) but reversed direction. (Duration must be measured only until the land command is seen).

[image]

ardugitter, 2022-08-17:
I'm going to close this because another "bug" I filed was invalid and may have the same user error as this one.

ardugitter, 2022-08-17:
Okay, having reviewed https://github.com/ArduPilot/ardupilot/issues/21471 which was invalid, I realise this bug appears to be valid. I have double checked the log with the GUIP filter and the acceleration targets are sent once per second as described:

[image]

Something is off about the behaviour of Arducopter as described in the top post. Thanks.

rmackay9, 2022-08-18:
I think I'm going to close this issue.  I hate to do this but I'm afraid that maybe you're chewing threw a bit too much ArduPilot developer time with these queries so it might be best if you investigate on your own for a couple more days and then if you're really sure you've found an issue feel free to re-open.

ardugitter, 2022-08-19:
By "chewing" do you mean wasting? I'm really sad if that is the case.

I'd like to apologise for the bug where I used the wrong accel/velocity compared to what I'd stated in the bug report. I had no idea you could filter by position target (or TBH even customise) the Mission Planner log analyser tables. I know that now. Such is the life of a relative noob who is learning.

The other bug report was due to a lack of understanding about the inner workings of the position controller (although I'd argue the behaviour of stopping accelerating once reaching a specified velocity made more sense from a user's point of view) and I think my bug reports are going to save you time if I understand that better so I'll try and unravel the complexities of that for myself.

That all said, I think this issue a genuine problem. The position controller always over-reacts to the first demand for a position target when it is just acceleration, or acceleration and velocity (when starting at a stand-still). I went out yesterday to get some real world data so I am going to post the information below to help those who come across this issue (and to allow me to follow up once I understand the controller code).

I am deeply appreciative of the excellent work being done on this project by the developers. :-)

ardugitter, 2022-08-19:
All versions both SITL and real are 4.2.2. The acceleration targets are in an odd direction because I wanted to fly perpendicular to the wind. The graphed velocity is just one of the axes for simplicity, even though the drone's velocity had components in northerly and easterly axes.

All of these logs/screenshots involve the following routine:

1) Takeoff
2) Set position target in direction D and wait 2 seconds.
3) Set position target in direction -D and wait 2 seconds.
4) Set position target in direction D and wait 2 seconds.
5) Set position target in direction -D and wait 2 seconds.
6) Land.

Acceleration in SITL:
[image]
Log: 
[2022-08-18 11-26-03-SITL.bin.zip](https://github.com/ArduPilot/ardupilot/files/9381134/2022-08-18.11-26-03-SITL.bin.zip)

Acceleration in a real drone:
[image]
Log:
[2022-08-18 17-13-33-real.bin.zip](https://github.com/ArduPilot/ardupilot/files/9381149/2022-08-18.17-13-33-real.bin.zip)

Velocity targets are reacted to more as I'd expect with the problem barely noticeable. SITL:
[image]
Log: 
[SITL-velocity.bin.zip](https://github.com/ArduPilot/ardupilot/files/9381198/SITL-velocity.bin.zip)

And velocity targets in a real drone:
[image]
Log:
[real-velocity.bin.zip](https://github.com/ArduPilot/ardupilot/files/9381223/real-velocity.bin.zip)

HOWEVER, the problem is seen when looking at setting VERTICAL velocity position targets (there is almost no downward motion at all when the first velocity target is upwards):
[image]
Log:
[velocity-vertical-real.bin.zip](https://github.com/ArduPilot/ardupilot/files/9381274/velocity-vertical-real.bin.zip)

I'm not expecting anyone to look at the logs except me, but I hope the photos cause people to keep this problem in the back of their minds regarding the position controller.

I will investigate myself and see not just why the controller over-reacts to the initial position target, but why it is so sluggish at the best of times compared to the controller that is used for manual flight (I assume they must be different given the latter is much more responsive to demands).

prathamesh7255, 2023-01-24:
@rmackay9 and @ardugitter I want to work on this issue and want some initial feedback on how to start with the issue.