ArduPilot/ardupilot #21470: Position targets that set both velocity and acceleration do not result in the velocity component being obeyed

ardugitter, 2022-08-17:
**Issue details**

I am trying to set both acceleration and velocity using the MAVLINK_MSG_ID_SET_POSITION_TARGET_GLOBAL_INT Mavlink message. This involves setting a typemask, 3 different velocity and 3 different acceleration fields.

I am finding that if I set velocity to 5m/s east and acceleration to 5m/s east, the SITL drone keeps accelerating and far exceeds the requested velocity.

This might be vaguely related to the other issue I just opened:
https://github.com/ArduPilot/ardupilot/issues/21469

**Version**
4.3.0dev SITL.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter SITL

**Hardware type**
SITL

**Logs**

This is the SITL log:

[2022-08-17 16-00-20.bin.zip](https://github.com/ArduPilot/ardupilot/files/9364007/2022-08-17.16-00-20.bin.zip)

This screenshot shows the velocity not being adhered to (I expect it to stop accelerating once the drone reaches 5m/s):

[image]

Thank you.

IamPete1, 2022-08-17:
It is not possible to maintain a constant 5m/s speed and constant 5m/s^2 acceleration, what your asking for is none physical. The message provides desired and velocity and desired acceleration. It is not desired velocity and acceleration limit.

ardugitter, 2022-08-17:
If this is the case, why is it possible to set both acceleration and velocity at the same time?

Arducopter is prioritising acceleration over velocity. It seems pointless to have a message that allows the user to specify both acceleration and velocity when only acceleration will be adhered to. 

I have limited understanding in this but I'd be grateful to understand the logic behind allowing both to be set at once.

IamPete1, 2022-08-17:
The vehicle can achieve a target acceleration and velocity at a single point in time, just not hold them both constant. Such control fine is useful if your trying to fly complex trajectories. For example the internal S-Curve controller gives a requested position, velocity and acceleration to the position controller each loop.

It is not prioritizing, it falls out of the velocity PID gains. They determine the acceleration the vehicle will will use to try and achieve the desired velocity. At some velocity error the acceleration that is used to achieve the desired velocity equals the desired acceleration your requesting and they cancel out resulting in a constant speed. You can test this by changing the velocity PIDs and see that steady state speed will also change.

If you want a constant velocity just provide velocity.

ardugitter, 2022-08-17:
Okay, in basic terms it sounds like the controller will change what it uses in its control loop between velocity and acceleration depending on things I don't understand (which is fine by me :-) ).

But just looking at it from a user's idealistic top level perspective, by specifying just an acceleration, I would expect the velocity vs time to be implicit. By specifying just velocity, then the acceleration would be implicit vs time (acceleration would be governed by some param presumably at the start, becoming zero once the velocity was reached).

Then as a user, I get confused thinking about setting both velocity and acceleration because like you say, they are both only possible at one moment in time. If I think less literally about what they mean and how they are parameters for a control loop, I can begin to understand why you'd set both. I'll close the "bug".

Thanks for the reply.

closed 2022-08-17 by ardugitter