ArduPilot/ardupilot #21471: Acceleration behaviour is different depending on which direction (NESW only) the position target points

ardugitter, 2022-08-17:
## Bug report

I realise I have filed 2 bug reports already today but I hope they are seen as valid. Here is another:

In SITL, if you request an acceleration-only position target using the MAVLINK_MSG_ID_SET_POSITION_TARGET_GLOBAL_INT Mavlink message, you will see different behaviour depending on if you fly north or fly east. 

In these tests, I ensured that the drone orientation was always north.

SIM_WIND_SPD is set to zero.

**Version**
4.3.0dev

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
SITL

**Logs**

The log contains 2 missions from takeoff to landing. 

[2022-08-17 16-42-36.bin.zip](https://github.com/ArduPilot/ardupilot/files/9364404/2022-08-17.16-42-36.bin.zip)

In the first, the drone is given 2 identical position targets to fly **east** at 7m/s/s. Each position target is 1 second apart. Following this, the reverse direction position target is given at the same interval and same total number of them (2). 

In the second, the drone is given 2 identical position targets to fly **north** at 7m/s/s. Each position target is 1 second apart. Following this, the reverse direction position target is given at the same interval and same total number of them (2). 

This image shows that reverse acceleration is applied correctly when flying east/west (first run, longitude) but the reverse acceleration is barely applied when flying north/south (second run, latitude).

[image]

IamPete1, 2022-08-17:
Your first command is asking for 7m/s Y velocity the second is asking for 7m/s^2 X acceleration. See the GUIP log message:

[image]

closed 2022-08-17 by IamPete1

ardugitter, 2022-08-17:
Sorry about that. User error. :-(