ArduPilot/ardupilot #21488: Copter: possible climb away during Auto mode Takeoff if Alt Frame is Terrain

rmackay9, 2022-08-19:
The takeoff controller exits early if it cannot get a terrain altitude ([see code here](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/takeoff.cpp#L114)) leaving the position controller without any updates which can lead to a climb away.

This can be reproduced in SITL by doing the following:

1. start SITL copter
2. param load ../Tools/autotest/default/copter-rangefinder.parm
3. restart SITL and execute step 2 again
4. create an Auto mission with a TAKEOFF command with Alt of at least 50m and Frame of AGL (aka "terrain")
[image]
5. param set AUTO_OPTIONS 3 (for covenience to allow arming and taking off in Auto)
6. auto
7. arm throttle

All is well until the vehicle climbs above the rangefinder's range (40m) at which point, "auto takeoff: failed to get terrain offset" will be displayed but the vehicle will keep climbing until the pilot retakes control in another mode.

[image]

This quite a serious bug and exists in the stable 4.2 release.  We should probably trigger a terrain failsafe if the vehicle is not landed.

peterbarker, 2022-08-19:
@rmackay9 why didn't the mission-terrain-prearm checks pick this up?  There's still the climb-away problem, but you shouldn't have been able to arm here, I think?

rmackay9, 2022-08-19:
@peterbarker,

I had to look into this but it seems that a terrain failsafe can never be active before the vehicle is armed.  Terrain failsafes only occur when the waypoint navigation library needs a terrain altitude for a waypoint or spline command and can't get it.

It also seems that if the terrain alt is not available when the takeoff command is started, it silently falls back to alt-above-home.

So really this failure only happens if the terrain alt is available as the takeoff starts but is later lost.

peterbarker, 2022-08-19:
We have pre-arm checks such that you should have the terrain around the waypoint items.

Ah.  I see the problem.  We only include waypoints, not takeoffs:

https://github.com/ardupilot/ardupilot/blob/pr%2Fsrv-channel-type-enum/libraries/AP_Terrain/TerrainMission.cpp#L75

rmackay9, 2022-08-20:
This has been resolved now by the PR linked above so closing.

closed 2022-08-20 by rmackay9