ArduPilot/ardupilot #21490: KakuteH7 Mini V1.3 not supported by latest firmware

developer-ken, 2022-08-19:
## Bug report

**Issue details**

I got a KakuteH7 Mini V1.1 and flashed it ardupilot. It worked well. I accidentally broke that board and bought a new one. That is a V1.3. I flashed it with the same version of ardupilot and it just won't work:

```
Config Error: INS: unable to initialise driver
Config Error: fix problem then reboot
```

**Version**
stable-4.2.2

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X frame, 4 rotor

**Hardware type**
KakuteH7 Mini V1.3 (version matters as some hardware was changed)

**Logs**
```
Config Error: INS: unable to initialise driver
Config Error: fix problem then reboot
```

developer-ken, 2022-08-19:
I changed the IMU from MPU6000 to BMI270 in hwdef.dat and the config errors disappear.  
But a new error:  
`PreArm: Logging failed`  
I think that can be addressed by just change somewhere in hwdef.dat.  
Is that something related to dataflash?
`SPIDEV dataflash SPI1 DEVID1 FLASH_CS   MODE3 32*MHZ 32*MHZ`
The new board has 1 Gib onboard flash, which is different from the old (v1.1) one.

andyp1per, 2022-08-19:
You need to flash the version labeled "KakuteH7Miniv2"

closed 2022-08-19 by andyp1per

developer-ken, 2022-08-19:
Oh I did'nt notice that.  
Flash KakuteH7Miniv2 firmware solved the problem.  
Thanks a lot.