ArduPilot/ardupilot #21513: We that need to improve Obstacle Avoidance Trajectory splicing

xianglunkai, 2022-08-22:
Modify AP_ Oapathplanner enables the planner to output a spatiotemporal trajectory instead of a point at a time, and each point carries absolute time. Every time mission_avoidance() is called in the avoidance interface, the track points should be obtained through time interpolation. In addition, the planning starting point should not always be the current position. In most cases, it should be the projection of the current position on the previous planning result (unless the control effect is poor or unexpected), and the time of high-speed movement of vehicles in the planning cycle should be considered.

xianglunkai, 2022-08-22:
I look forward to discussion. I have some ideas, but they are not good enough.

rmackay9, 2022-08-24:
I agree very much with this point, "projection of the current position on the previous planning result".

xianglunkai, 2022-08-24:
AI cameras are really good, but the distance is limited.  I think we can try to use  radar, such as ars408-can firstly.

rmackay9, 2022-08-24:
@xianglunkai,

Yes, radar is good.  By the way, I did a presentation a couple of weeks ago about boats and object avoidance. I don't have a recording of the presentation but [here on the slides](https://docs.google.com/presentation/d/1g7fw09K4A_irCV7D_y7ot6ZwbRZUpdtTpa_mwMNOlMo/edit?usp=sharing).  It's not much and it's quite similar to my AP dev conference presentation but..

xianglunkai, 2022-08-24:
Thank you very much!

timtuxworth, 2022-08-24:
This could apply just as much to fixed objects. In plane I would like to see the plane able to avoid a known/pre-configured geofence instead of just plowing blindly into it. It could also apply to ADSB detected vehicles where the location and even velocity are known with certainty so the plane doesn't have to detect anything, just avoid something it knows with certainty is (or will be) in the way.