ArduPilot/ardupilot #21534: get_next_nav_cmd does not correctly account for do_jump commands

mentonin, 2022-08-24:
## Bug report

**Issue details**

`AP_Mission::get_next_nav_cmd` does not correctly account for `do_jump` commands, even though it says it does. It gets the next command using `get_next_cmd`, which does account for `do_jump`, but then continues the search through simple index incrementing. If the command list is something like

1. non-nav command
2. nav command
3. do-jump to 1
4. nav command

`get_next_nav_cmd(3)` will return command 4, instead of 2

**Version**
master branch

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

gcs-sovps, 2026-01-22:
Bumping this. We found this when doing a LOITER_TO_ALT with a tangent exit, followed by a DO_JUMP_TAG. It causes the loiter exit to be disregarded (tangential or normal) and the vehicle to stop it's loiter early and do a sharp turn which is undesirable.

rmackay9, 2026-11-10:
I believe this issue has been resolved by these two PRs so I will close this issue.  The fixes will be included in 4.8.

- https://github.com/ArduPilot/ardupilot/pull/35049
- https://github.com/ArduPilot/ardupilot/pull/35041

closed 2026-11-10 by rmackay9