ArduPilot/ardupilot #21538: Tools: install scripts for ubuntu need to include instlal of dronecan

rmackay9, 2022-08-25:
New installs apparently need the python "dronecan" package to be installed.  This should be included in the environment install scripts for ubuntu and perhaps other environments as well.

https://github.com/ArduPilot/ardupilot/blob/master/Tools/environment_install/install-prereqs-ubuntu.sh

khancyr, 2022-08-25:
What were you trying to build ? 
It looks like our env tester do catch this

rmackay9, 2022-08-25:
@khancyr,

I was running this on my Windows11 machine's WSL2 environment.

khancyr, 2022-08-25:
I don't have dronecan and it is working fine on my side to build SITL.
What target you are building ? SITL ?
We should surely add dronecan but I would like to know what lead to trigger this error to add it to test suite

rmackay9, 2022-08-25:
@khancyr,

txs. I was running sim_vehicle.sh..

khancyr, 2022-08-25:
I cannot reproduce with `sim_vehicle.py -v ArduCopter `

rmackay9, 2022-08-25:
@khancyr, Perhaps it only happens with a fresh install?  .. or a distclean?  I'm not sure but the issue certainly happened because the evidence is above.

khancyr, 2022-08-25:
ok we need to add it. I manage to reproduce on SITL. 
AP_OpenDroneID support make it mandatory.

closed 2022-08-26 by tridge

LouisRenWeiWei, 2023-01-19:
dronecan==1.0.8