ArduPilot/ardupilot #21547: AP_Mount: angle limits held in MNT_ANGMIN/MAX_xxx may not always be respected

rmackay9, 2022-08-26:
In some (or more likely all) of our gimbal drivers the angle limits input by the user (MTN_ANGMIN/MAX_ROLL/TIL/PAN) may not be respected.

This can be seen in the screen shot below of a SITL test in which the SToRM32 mavlink gimbal driver was enabled (MNT_TYPE = 4) and MNT_ANGMIN/MAX_PAN were set to 0).  Despite this the debug output at the bottom in red shows that non-zero yaw targets were being sent to the gimbal as a DO_SET_ROI command was being executed.
[image]

Tangoviktor, 2022-11-14:
@rmackay9 How can this limit angle be solved? A lot of users have hard limits on their gimbals and without having the possibility of using soft limits in code then at least with storm32 the gimbal imu floats away. I have seen this couple of times when the gimbal is hitting the limiter during flight and after a while the imu data becomes garbage.

rmackay9, 2022-12-19:
There is a duplicate issue here https://github.com/ArduPilot/ardupilot/issues/13815