ArduPilot/ardupilot #21562: Web Tool for FFT Plots to be used Harmonic Notch Filtering

hendjoshsr71, 2022-08-29:
A web tool similar to [https://firmware.ardupilot.org/Tools/FilterTool/](https://firmware.ardupilot.org/Tools/FilterTool/) for FFT analysis should be created to allow for easy platform agnostic analysis.

**Things it should be able to do **
- Plot RATE.Outs  & RC_Inputs  and allow selection by time period (via zooming) 
- Plot FFT pre & post filters 
- Scale selector for log vs linear

Bonus points
For letting you select some points and save those by frequency and dB. 

Code for the Notch Filter Tool: [here](https://github.com/ArduPilot/ardupilot/tree/master/Tools/FilterTestTool)

**Code for current FFT Tools**
mavfft_isbd.py : [here](https://github.com/ArduPilot/pymavlink/blob/master/tools/mavfft_isb.py)
MAVExplorer FFT tool: [here](https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/lib/mav_fft.py)

ES-Alexander, 2022-08-29:
@Williangalvani would it make sense to add this to the existing log viewer?

Since FFT is another way of looking at data, perhaps there could be an extra set of axes for FFTs, that each displayed signal could optionally be shown on (e.g. via a per-signal checkbox to 'show fft', and possibly a similar checkbox to 'show over time' in case someone wants only FFT for some signals). 

Frequency focused filtering (like the existing `lowpass` function, and the ones in the FilterTool) still just creates another time-based signal, so should be easy to show as before and after filtering by just having the base signal as one element and the filtered signal as another one. The `RATE.Outs` and `RC_Inputs` signals are already plottable, and time-period selection via zooming is already implemented.

hendjoshsr71, 2022-08-29:
Fixed the link to the notch code in the original post .

I agree it would make a great addition to plot.ardupilot.org ! Plus I think it would be nice if we could not have the same tool in 5 different places. (Like we do now...)

To Peter's point it would be nice to create a common library for such tools.   And to think about how to make it easier for java scripts noobs (**me**) to help. 

I love of MAVExplorer but it breaks far too often on windows I can't tell partners to use it unless they run WSL2.

Williangalvani, 2022-08-29:
Hey guys.

I'm definitely interested in expanding the functionality in logviewer.
Meanwhile, I quickly stripped things out of it so others can build on top of it.
The async nature of the parser makes it a bit weird to use, so I may do a promise wrapper around it later. That's what I have in logviewer IIRC.

Williangalvani, 2022-08-29:
And I forgot to link it =|
https://github.com/Williangalvani/JsDataflashParser