ArduPilot/ardupilot #21572: AC_PrecLand: remove PLND_YAW_ALIGN parameter

rmackay9, 2022-08-30:
PR https://github.com/ArduPilot/ardupilot/pull/21132 adds a new PLND_ORIENT parameter.  We should remove the existing PLND_YAW_ALIGN parameter because otherwise users may be confused about which to set.

This will lead to a slight loss of functionality for Copter users who will no longer be able to specify the downward facing camera's yaw orientation in degrees.  Instead they will only be about to choose from eight different yaw orientations (45deg, 90deg, etc). but this should be acceptable.

chobitsfan, 2022-09-30:
Hi @rmackay9 I think maybe I could help to fix this?

rmackay9, 2022-09-30:
@chobitsfan,

That would be great, thanks!

chobitsfan, 2022-10-04:
Hi @rmackay9
@rishabsingh3003 point out in https://github.com/ArduPilot/ardupilot/pull/21849#pullrequestreview-1127626926 that  PLND_ORIENT is a Rover-only param. Are we going to make it available to copter? Thank you