ArduPilot/ardupilot #21584: method for unlocking signed bootloaders

tridge, 2022-08-31:
For vehicles that are originally sold in the US then are re-sold in another country that doesn't require remote ID it would be nice to allow a way for a vendor to unlock the bootloader remotely.
I think the following would work:
 - end user first convinces the vendor that they can legitimately fly without RemoteID. This may involve contractual/legal work that is out of the scope of this proposal
 - we could include in ROMFS both the secure boot bootloader and the base bootloader that doesn't have secure boot enabled (we could call this @ROMFS/bootloader_unsigned.bin)
 - end user would forward mavlink to the vendors support engineer
 - support engineer would confirm vehicles location with GPS and confirm it can legally fly without RemoteID
 - support engineer would open a plugin screen in MissionPlanner for unlocking boards
 - in that screen the support engineer would point to the vendors private key 
 - the plugin would get the boards STM32 unique ID from MAVLink connection
 - the plugin would sign the STM32 unique ID, creating a signed blob
 - that signed blob would be sent to the vehicle over the forwarded mavlink connection using a new mavlink pkt for this purpose
 - the vehicle when it gets that command it would use the public keys from the bootloader to check the signing of the STM32 ID is correct
 - if the signing is correct then it will flash the unsigned bootloader

Note that this technique does not open a hole that impacts anything beyond that single vehicle

CraigElder, 2022-08-31:
This sounds like a very good idea

timtuxworth, 2022-08-31:
> * the vehicle when it gets that command it would use the public keys from the bootloader to check the signing of the STM32 ID is correct
> * if the signing is correct then it will flash the unsigned bootloader

This all sounds good except for these two lines which are a bit vague. 
* What does "the vehicle" mean. Is this ArduPilot running on the FC? Is it the secure boot loader? 
* what does "it will flash the unsigned boot loader" mean? what is "it"?
* Where does the unsigned base boot loader come into it? (its mentioned at the beginning but not referenced anywhere)
* What is to stop a bad actor from running the "unsigned boot loader" to install their own boot loader and/or their own build of ArduPilot?

peterbarker, 2022-08-31:
> * we could include in ROMFS both the secure boot bootloader and the base bootloader that doesn't have secure boot enabled (we could call this @ROMFS/bootloader_unsigned.bin)

This could also be transferred via FTP to an SD card on the autopilot - in such a form that the ArduPilot firmware would be able to verify that it's been signed by a relevant key.  Two advantages I can think of - one is space and the second I won't mention here.

tridge, 2022-10-01:
merged

closed 2022-10-01 by tridge