ArduPilot/ardupilot #21649: Latest firmware won't work with the Matek F405 WTE Flight Controller

Isaac542, 2022-09-06:
## Bug report

**Issue details**
I am flashing the Matek F405 TE firmware, which is recommended by the manufacturer. I can connect to the board right after flashing but if I disconnect it from USB and plug it back in the board is not recognized by my computer. There is no issue with the board it works just fine with INav firmware.

**Version**
4.2.3
I have also tried other versions with no luck.
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Vtol
**Hardware type**
Matek F405 WTE
**Logs**

adalta87, 2022-09-06:
Did you flash thr firmware with the bootloader usually xxxx_bl.hex

Isaac542, 2022-09-06:
I was using the hex file with no bootloader before. I tried it and it works with the bootloader file. Thanks for the help.

closed 2022-09-06 by IamPete1