ArduPilot/ardupilot #21684: Support for Omnibus F4 V3S Plus Flight Control

devanesanandrews, 2022-09-10:
Support for Omnibus F4 V3S Plus Flight Control 

Specifications:
Board Dimensions: 36x36mm
Montagegaten: 30.5×30.5mm
STM32 F405 MCU
Dshot protocol
Slepen en Neerzetten OSD geconfigureerd via Betaflight
BMI270 6 axis SPI gyro & accelerometer
SBUS / PPM EN Spketrum DSMX poorten
MicroSD BlackBox
Baro (BMP280)
5v3a 9V3A BEC
On-board video filters (alleen 5V naar VTX en camera)

it uses BMI270 6 axis SPI gyro & accelerometer

**Platform**
[X] All
[X] AntennaTracker
[X] Copter
[X ] Plane
[X ] Rover

[image]

devanesanandrews, 2022-09-12:
[omnibusf4pro.zip](https://github.com/ArduPilot/ardupilot/files/9548094/omnibusf4pro.zip)

i modified the hwdef file of omnibusf4 pro and it works
i havent tested in a drone, but
SPIDEV bmi270    SPI1 DEVID2 MPU6000_CS   MODE3   1*MHZ  10*MHZ # Clock is 100Mhz so highest clock <= 10Mhz is 100Mhz/16

i added this line from mamabah74v3 is the clock speed right

andyp1per, 2022-09-12:
> is the clock speed right

Probably good enough. This sensor won't go beyond 10Mhz anyway

devanesanandrews, 2022-09-12:
@andyp1per i am not aware of the the modes used, is what i used is correct, if yes then, i can create a pull request and add this to the list

Thank you

andyp1per, 2022-09-13:
> not aware of the the modes used

This is just a clocking thing - we generally always use mode 3

andypnz, 2022-09-19:
I built firmware for plane using your hwdef - thank you! It installs fine. Accelerometer calibration is good, and the AH shows correct. Serial ports are working.

But compass calibration fails - my BN-880 compass is correctly listed (HMC5883) - but the calibration just keeps starting again... I tried another GSP/Compass and same issue.

Any ideas on trouble shooting? I have posted here as I wonder if it is related to the hwdef BMI270 setup....

andyp1per, 2022-09-19:
If the compass is detected then likely an orientation problem

andyp1per, 2022-09-19:
@devanesanandrews can you do a PR for this hwdef?

devanesanandrews, 2022-09-19:
@andyp1per actually im not good at using Github, but i ll try and do it today

Thank you

tridge, 2022-09-19:
@devanesanandrews if its just a different IMU then we can support both IMUs on the current build by just adding a line for it

devanesanandrews, 2022-09-19:
@tridge @andyp1per 

i have requested a pull request

https://github.com/ArduPilot/ardupilot/pull/21750#issue-1378401349

lazyan, 2022-10-08:
> [omnibusf4pro.zip](https://github.com/ArduPilot/ardupilot/files/9548094/omnibusf4pro.zip)
>
> i modified the hwdef file of omnibusf4 pro and it works i havent tested in a drone, but SPIDEV bmi270 SPI1 [...]

I've made almost the same, but with the different orientation: ROTATION_ROLL_180_YAW_270. These orientation parameters are checked in Mission Planner and correct. GPS/compass/RC are accessible, acc/compass calibtations are OK.
Maybe there are some variants of the boards with different bmi270 orientations. mine is labelled as omnibus F4 V3S plus with bmi270 in dark brown color.
Anyway I confirm hwdef files are OK.

bitopu, 2023-03-03:
do you have the manual for this flight controller?

teho03, 2023-03-05:
Is it safe to have USB connected at the same time as the Lipo3S?

lazyan, 2023-03-05:
I've tried with lion 1s without problem. But I think no problem with other
options...

lazyan, 2023-03-05:
No. The only information I have is from this forum

teho03, 2023-03-05:
Thank you Lazyan for the reply.    There seems to be very little information on this FC on the web and I've certainly not found a manual.   I had assumed that since it could be fed directly from the Lipo and that it had an on-board 5V suply, there would be no conflict when feeding direct from battery and USB, but I wanted to make sure, so I had to ask.  Many thanks.

andyp1per, 2023-03-07:
It's quite unfortunate that there are different orientations - there is no way of easily dealing with that

git500, 2023-09-14:
Is any body has a pinount (schema) for this board F4V3S plus? I'm looking for SWD/SWC ports...

anuragbhatt1805, 2024-04-07:
> I built firmware for plane using your hwdef - thank you! It installs fine. Accelerometer calibration is good, and the AH shows correct. Serial ports are working.
>
> But compass calibration fails - my [...]

how serial ports got connected??

devanesanandrews, 2024-04-07:
Omnibus F4 V3S

has got an issue, it comes with multiple variants, some with bmi270 and some with mpu6000. For now i am using Omnibusf4 firmware since it has all the sensors and orientation might be an issue so you have to change the ahrs orientation. Serial ports are same as the omnibus f4 pro v2, but the Rx6 is connected to PPM, you can change the BRD with respective values for appropriate connection

devanesanandrews, 2024-04-07:
@tridge @tridge 

this board has an issue since it comes with multiple variants with bunch of diffirent sensors and orientation. I dont think so we can use a dedicated firmware because of this. 

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/omnibusf4pro/hwdef.dat

has all the sensors included, i am using this firmware. It can be configured properly and used. 
So i would close this issue

closed 2024-04-07 by devanesanandrews

Mistichoy, 2025-08-30:
hola como están mi consulta es sobre esta controladora la actualice en inav y ahora no me anda el giróscopo me pasarían que firmware puedo usar saludos