ArduPilot/ardupilot #21722: Tools: install-ROS-ubuntu.sh failing due to robot_pose_publisher dependency

rmackay9, 2022-09-14:
The [install-ROS-ubuntu.sh](https://github.com/ArduPilot/ardupilot/blob/master/Tools/environment_install/install-ROS-ubuntu.sh) script fails with the following message:

```
ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies:
ardupilot_ros: Cannot locate rosdep definition for [robot_pose_publisher]
```

The issue is probably actually in the [ardupilot_ros repo](https://github.com/ArduPilot/ardupilot_ros), perhaps caused by [this commit](https://github.com/ArduPilot/ardupilot_ros/commit/f23bfd8ca46a5c49b96651972e9392c808ab6df1).

snktshrma, 2022-09-14:
Hello @rmackay9 sir!
Just check this and found out robot_pose_publisher is available only upto ros kinetic! My bad that I added it as a dependency for noetic and melodic.
Probably you can clone git clone `https://github.com/GT-RAIL/robot_pose_publisher.git` repo in ros_ws/src directory!
I'll update the package as well as the ibnstall-ROS-ubuntu.sh script

snktshrma, 2022-09-14:
https://github.com/ArduPilot/ardupilot_ros/pull/3

rmackay9, 2022-09-14:
@snktshrma thanks for that.  As mentioned in another discussion, the script is now failing as it looks for cartographer_ros

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ardupilot_ros: Cannot locate rosdep definition for [cartographer_ros]

snktshrma, 2022-09-15:
https://github.com/ArduPilot/ardupilot_ros/pull/4/files

snktshrma, 2022-09-15:
Sir just to inform these instructions are not yet tested on the real environment!