ArduPilot/ardupilot #21724: Request for desired speed

sulakshanarya, 2022-09-14:
## Feature request
Is there a way to request the desired speed from the flight controller after the DO_CHANGE_SPEED changes the wp speed.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter-4.x
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
I am migrating from Copter-3.6.6 to Copter-4.2 and in Copter-3.6.6. In Copter-3.6.6 I used request for the WPNAV_SPEED parameter value for desired speed. But Copter-4.2 is no more updating it when changed from DO_CHANGE_SPEED.

rmackay9, 2022-09-16:
I think the only possibilities are via the [POSITION_TARGET_GLOBAL_INT](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L4523) or [POSITION_TARGET_LOCAL_NED](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L4487).

AP does not currently fill in the target velocities for the POSITION_TARGET_GLOBAL_INT message but it could.  FYI @hendjoshsr71.

sulakshanarya, 2022-09-28:
Thank you @rmackay9. I spent here sometime to understand if the suggested implementation works. What I see is the flight controller publish [POSITION_TARGET_GLOBAL_INT](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L4523) only target is set/available. But the [set_speed_xy](https://github.com/ArduPilot/ardupilot/blob/aa757501f43dd4dd966aa7d44cde1c554cfe93bd/libraries/AC_WPNav/AC_WPNav.cpp#L209) still accepts the user change speed command (desired speed) even when target is not set, for example even when drone disarmed and this will confuse the user since desired speed is not updated in GCS. 

Or do you see a better implementation for it @hendjoshsr71 ?

sulakshanarya, 2022-09-28:
Thank  you @rmackay9. The flight controller streams [POSITION_TARGET_GLOBAL_INT](https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L4523) only when target is set, but accept the DO_CHANGE_SPEED command even when target is not set. This would confuse the user if desired speed is changed but not visible at GCS. 

Or is there a better way to implement?