ArduPilot/ardupilot #21792: Diatone Mamba F405 MK4 icm42688 support for MambaF405-2022 board

P-I-Engineer, 2022-09-23:
Please add target for mambaF4052022B. 

Mamba F405 mk4 board firmware does not have a driver for the the alternate version of the board that is populated with a ICM42688P

in BF/INAV this is handled by having alternate target MAMBAF405_2022B
https://github.com/betaflight/unified-targets/blob/master/configs/default/DIAT-MAMBAF405_2022B.config

https://www.diatone.us/products/mb-mk4-f405-mini-fc

thank you

vidma, 2022-12-02:
i think it could be based on this one https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_HAL_ChibiOS/hwdef/MambaF405-2022 by changing a few things according to BF/INAV.

seem to boot partially ok with original `MambaF405-2022` (visible in QGC) , but doesn't work fully (always complain `Config Error: fix problem then reboot repeatedly.`).

andyp1per, 2022-12-02:
Can you try this?

[arducopter.zip](https://github.com/ArduPilot/ardupilot/files/10140009/arducopter.zip)

vidma, 2022-12-02:
@andyp1per do you have arduplane build too, or  evenbetter source code patch/diff / branch?

andyp1per, 2022-12-02:
https://github.com/ArduPilot/ardupilot/pull/22332

vidma, 2022-12-02:
> #22332

ok,  @andyp1per ,  seem to help a bit (IMU possibly fixed), still getting issue with BARO both with my arduplane build and your your `apj` :

[image]

---

p.s. is this correct way to build? do i need explicit call to `./Tools/scripts/build_bootloaders.py` ?

```bash
export PATH="/opt/gcc-arm-none-eabi-10-2020-q4-major/bin:$PATH"
./Tools/scripts/build_bootloaders.py MambaF405-2022
./waf configure --board MambaF405-2022
./waf plane

ls ./build/MambaF405-2022/bin/
# arduplane  arduplane.apj  arduplane.bin  arduplane.hex  arduplane_with_bl.hex
```

andyp1per, 2022-12-02:
You don't need to rebuild the bootloader - there is a binary version already and nothing has changed in rev b that would make a bootloader rebuild necessary. Other than that looks ok. LMK when you figure out the correct orientation for the IMU

vidma, 2022-12-03:
@andyp1per here, seem like i fixed the IMU orientation too for 2022B (hopefully, in GQC pitch, roll, and "compass[yaw]" movements seem to match real movements of https://www.diatone.us/collections/flightcontroller/products/mamba-mk4-f405mini-flight-control-20mm-m2  ), take this PR https://github.com/andyp1per/ardupilot/pull/29

closed 2022-12-06 by tridge