ArduPilot/ardupilot #21804: Raspberry Pi Cannot Obtain GPS Information from Pixhawk 6C + DroneKit + ArduRover stable-4.2.3

jakobjackson2016, 2022-09-26:
## Bug report

**Issue details**

_Unable to pull (obtain) GPS lat/long info on Raspberry Pi (RPi) using DroneKit SDK's API: `vehicle.location.global_relative_frame` from HolyBro Pixhawk 6C.  Exact same `.py` script to pull this GPS info on RPi with exact same firmware, GPS and other related setup parameters work with HolyBro Pixhawk 4 + RPi but it DOES NOT work with Pixhawk 6C + RPi. I can obtain `vehicle.mode` info from Pixhawk 6C on RPi but not `vehicle.location.global_relative_frame` info which provides GPS lat/long information on RPi from Pixhawk 6C; the output for lat/long remains blank. There seems to be a bug preventing this API to pull lat/long info from Pix6C with rover-stable-4.2.3 firmware. Just FYI: I'm upgrading my rover from HolyBro Pix4 to Pix6C. RPi <-> Pix6C connection setup: USB to micro-USB._

**Version**
_ArduRover stable-4.2.3_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

**Airframe type**
_X_

**Hardware type**
_Pixhawk 6C_

**Logs**
_[https://drive.google.com/file/d/1MATgTFCmkF_j4VDdLHi9faw6CzzSPPUV/view?usp=sharing](https://drive.google.com/file/d/1MATgTFCmkF_j4VDdLHi9faw6CzzSPPUV/view?usp=sharing)_

LeighDavis, 2022-09-27:
Looking at the logs, I simulated and tested OP's issue with my hardware here and I am able to recreate and observe it. Looks like a bug but I am not quite sure what the issue is here. Maybe the devs will be able to troubleshoot this? @rmackay9 Can you please further assist with this issue?

closed 2022-09-27 by jakobjackson2016

jakobjackson2016, 2022-09-28:
So I was able to figure out the root cause of the problem. I am able to read all eight RC channel values on Pixhawk4 but not on Pixhawk6C. My Tx is FrSky QX7 and Rx is FrSky R10Pro. My algorithm is based on the 8 switch’s channel values and that is the issue here. Cannot obtain channel values on RPi. Getting: `{‘1’: None, ‘2’: None, … , ‘8’: None}`. Not sure why I am not able to obtain channel values on RPi via Pix6C only.