ArduPilot/ardupilot #21817: Getting "Sending unknown message (50) " continously

Ltomi78, 2022-09-27:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**
I ve got information from Randy Mackay the reason is there is a coding error in this line in custom fw.
https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L5752
_Please describe the problem_
I upload 4.3.0 custom build fw https://custom.ardupilot.org/ and now getting this message continously: 
"Sending Unknown message (50)"
Setup is: Fc matek f405 wing + gps and compass
**Version**
4.3.0 build with custom fw builder for matek f405 wing target 

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
flying wing

**Hardware type**
_What autopilot hardware was used? 
Matek F405 wing

**Logs**
_Please provide a link to any relevant logs that show the issue_
[image]
https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L5752

macfly1202, 2022-09-28:
I have this error on MATEK F405 target too, with custom firmware ArduPlane V4.3.0dev (f9bc7e44). 
Displayed on DJI FPV OSD. 

[image]

IamPete1, 2022-09-28:
Fixed by https://github.com/ArduPilot/ardupilot/pull/21828

closed 2022-09-28 by IamPete1

Ltomi78, 2022-09-28:
> I have this error on MATEK F405 target too, with custom firmware ArduPlane V4.3.0dev ([f9bc7e4](https://github.com/ArduPilot/ardupilot/commit/f9bc7e44f85266abd66fe59a93d0e75de3a7b645)). Displayed on DJI FPV OSD.
> 
> [image]

 Today the devs fixed it...you can build a 4.4.0 dev from the builder without this issue😉
[image]