ArduPilot/ardupilot #21819: GPS not auto detected, probably only when CRSF frame problems happen

krzysztofkuczek, 2022-09-27:
## Bug report

The uBlox GPS reciver is not detected, when one of the usart port is set to be used as rc input for CRSF compatible (not fully compatible) reciver.

The setup is like following
SERIAL1_PROTOCOL = 5 (GPS connected)
SERIAL3_PROTOCOL = 23 (QLRS CRSF compatible RX connected)
SERIAL4_PROTOCOL = 42 (DJI Air Module connected)

Then after the reboot, the GPS is not detected. The RX does not send any frame, so it seems to be not discovered which blocks GPS detection ? Even when RX is powered, due to not fully implemented CRSF proto the log shows "CRSFv2: requesting RX device info", but GPS is not detected. The RX works fine, channels are correctly read.

The work around is to set any other seriall port to 23 it solve the problem, everything works fine 
So
SERIAL5_PROTOCOL = 23 (nothing is connected to the port)
And everything works fine (GPS 1: detected as u-blox at 230400 baud)

The question is, if it's related to old revo mini board, or it's some problem with CRSF proto implementation, which could potentially block GPS connectivity...

**Version**
ArduPlane V4.2.3 (55685c43), revo-mini-i2c 00430023 3235510C 3336363

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
mini glider

**Hardware type**
revo-mini

**Logs**
N/A

andyp1per, 2022-09-27:
Most likely a DMA issue

krzysztofkuczek, 2022-09-27:
Ok, I think it's related only to the old revo board. Can close it, no one will investigate it ...

closed 2022-09-27 by krzysztofkuczek