ArduPilot/ardupilot #21866: COPTER: Adding RTL to SMARTRTL

muramura, 2022-10-03:
## Feature request

**Is your feature request related to a problem? Please describe.**

I made a random flight with SMARTRTL.
I also flew near the takeoff point.
I selected "SMARTRTL or RTL" for the battery fail action.
The vehicle triggered the battery fail safe.
This vehicle passed the takeoff point.

EX:
[image]

**Describe the solution you'd like**

I want to land the vehicle when it approaches the takeoff point while in reverse.

**Describe alternatives you've considered**

I would put the RTL region at the takeoff point.
If the vehicle enters the RTL area while in reverse in SMARTRTL, switch the flight mode to RTL.
I think this change would cover flight trajectories that do not intersect.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

rmackay9, 2022-10-03:
I guess it didn't detect the loop because the altitude was different?

muramura, 2022-10-07:
@rmackay9 san. I accept.