ArduPilot/ardupilot #21880: Mavlink command MAV_CMD_DO_SET_MODE with parameter MAV_MODE_FLAG_DECODE_POSITION_SAFETY should not disable the motors when the quadcopter is in flight

ardugitter, 2022-10-04:
## Feature request

**Is your feature request related to a problem? Please describe.**

I've listed this as a feature request, but in my opinion it is a bug. I just didn't want to use up the time of developers so please raise it to a bug if you think it should be.

**Describe the solution you'd like**
If you attempt to disarm the hardware safety switch in software by sending the MAV_CMD_DO_SET_MODE command with param1 as  MAV_MODE_FLAG_DECODE_POSITION_SAFETY and param2 as 0, this should have no effect if the drone is in flight. At present, it will disable all motors and the drone will crash.

**Describe alternatives you've considered**
The MAV_CMD_COMPONENT_ARM_DISARM command is ignored if the drone is flying *unless* you set a magic number. It would be nice if the same applied in this situation as param3 is available for this purpose.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
I crashed my drone due to disarming the hardware safety switch mid-air and it would be nice if this were impossible for anyone else to do. There is no reason for disarming the safety switch while flying except for malicious reasons of hacking.

Thanks.

IamPete1, 2022-10-04:
This is safety critical, we cant can change it, this behavior will be included in safety cases. Much better that we stop the motors when told to than ignore the command and fly away or into someone. This command is quite hard to send by mistake, unlike disarm which is often a big button on the GCS.

ardugitter, 2022-10-04:
You already have the disarm-with-magic-number if you want to kill the motors instantly. This logically (based on the name) seems to be the operation to use to kill the motors.

But I realise somewhere down the line a decision was made that the safety switch too should be allowed to kill the motors. Thanks for the reply!

closed 2022-10-04 by ardugitter