ArduPilot/ardupilot #21897: Vision Position Estimate not working in ArduRover

harshmittal2210, 2022-10-06:
## Bug report
I am using [vision_to_mavros/t265_to_mavlink.py at master · thien94/vision_to_mavros · GitHub](https://github.com/thien94/vision_to_mavros/blob/master/scripts/t265_to_mavlink.py) for position estimates. For the initial few seconds, I am able to see the packets but after that VISION_POSITION_ESTIMATES frequency decreases to 0.

I have added all the parameters mentioned here: [Intel RealSense T265 — Copter documentation](https://ardupilot.org/copter/docs/common-vio-tracking-camera.html)

Just the VISO_TYPE is not there in the ArduRover Parameter list.

Does rover require some additional parameters or this feature is not yet supported?

**Version**
4.3

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ x ] Rover
[  ] Submarine

**Airframe type**
ArduRover

**Hardware type**
Pixhawk 4

IamPete1, 2022-10-06:
`VISO_TYPE` should be there unless there unless you have disabled it. You could try loading the latest firmware.

harshmittal2210, 2022-10-06:
I flashed the latest firmware only, I'll try again to check if I missed something.

rmackay9, 2022-10-06:
the VISO_TYPE parameter certainly should be there. The only reason I can imagine it wouldn't be there is because the feature has been disabled for the Pixhawk4 Rover firmware but I just did a test and it was included.

Maybe double check that 4.3 is really being loaded on the vehicle.  This should be displayed in MP's messages tab.

I think I might close this issue because it's really a support call that should go into [the Rover forums ](https://discuss.ardupilot.org/c/ardurover/rover43/168) first.  I'll still see replies here.  Sorry to be slightly strict but if we don't our to-do list turns into another support forum.

closed 2022-10-06 by rmackay9