ArduPilot/ardupilot #21903: Smart RTL with only few waypoints for a arduplane.

ninja-zx11, 2022-10-06:
## Feature request

I would like to initiate a feature request of implementing just few points for smart RTL (like a copter) on arduplane. 

Lets say you are flying at around 300mts over the passage between the high mountains ( 2000m) and you accidently got behind the mountain. If using terrain data, plane would loose all its battery to achieve 2000+ meters. But if there is a smart RTL, it would quickly follow the path back.

I am aware of the memory limitation on the boards but if we can get only few points for smart RTL, it would be very useful.

May be only  few points would be useful  just to come back LOS.
**First few points for smart RTL and then switch to normal RTL like copter**.  In that way you can never loose your plane if accidently flying behind a high mountain.

**Platform**

[ * ] Plane

timtuxworth, 2022-10-06:
I have personally had several crashes, and destroyed 2 planes, when RTL decided to fly through or outside a clearly marked geofence in order to fly home after a "failsafe". I have marked geofences around mountains, buildings, physical fences and trees. Fixed wing RTL will fly right into them. A smart RTL would be a good start to addressing this.

ninja-zx11, 2022-10-07:
Exactly. I crashed my nicely built Mini Talon last month because of that:(

cbf123, 2024-05-30:
I think this is going to be very similar to https://github.com/ArduPilot/ardupilot/issues/2380

I don't think memory is necessarily a limitation since you already need a decent chunk of memory to handle terrain data in the first place.