ArduPilot/ardupilot #21920: GSOC: Support twisting-blade rotor control

peterbarker, 2022-10-08:
## Feature request

Achieiving roll and pitch movement by using acceleration to twist blades is a technique used for a long time in small UAVs.

Tom Stanton has done a series of videos implementing this - e.g. https://www.youtube.com/watch?v=55G-mwT9uCE

**Describe the solution you'd like**

An ArduPilot implementation!

```
BillGeyer: @Peter Barker Probably best to add this to the copter vehicle type as opposed to Tradheli since it still uses varying rpm for thrust. However you could probably make use of the swashplate library but would need some modification as it provides fixed frame inputs and this would require rotating frame inputs for the motors. I would be will to help mentor this project. 
```

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

bnsgeyer, 2022-10-08:
This project could also include incorporating a fixed pitch helicopter frame into copter.  A fixed pitch rotor helicopter can be flown using the Tradheli vehicle type but it requires some special setup and doesn't account for the nonlinear thrust with rpm.

007harshmahajan, 2022-12-08:
I want to work on this feature so, did some research and found out that this research paper belong to Jimmy Paulos, he has published a video talking about the research paper but finally when I was trying to get the research paper I found that, the paper is patented 

- [Jimmy paulos website](https://www.grasp.upenn.edu/people/jimmy-paulos/)
- [Jimmy's video](www.youtube.com/watch?v=aEPf0QHVuMM&t=88s)
- [Patent](https://patents.google.com/patent/US20160001877)