ArduPilot/ardupilot #21921: Copter: RC not found pre-arm if only RC_OVERRIDES are used

rmackay9, 2022-10-08:
Testing latest Cotper-4.4.0-dev I've found that if only a joystick is used (from MP, aka RC_OVERRIDES) then a pre-arm failure appears, "PreArm: RC not found".  I can see the RC values changing in MP's Setup >> RC Calibration screen and RC failsafes are triggered or cleared as the joystick is enabled or disabled.

[image]

[image]

muramura, 2022-10-08:
This error occurs when the radio signal is not received regularly.
This can help detect a problem with the radio's receiver.
MAVLINK communication is OK within 5 seconds, assuming there is a delay.
When doing RC_OVERRIDE with MAVLINK communication, I think it is necessary to assume this delay in the task ('output' the same value as the previous one until a new signal arrives).

rmackay9, 2022-10-08:
@muramura,

thanks for the feedback but this is not the issue actually.  This is a real bug because the new check added in this PR https://github.com/ArduPilot/ardupilot/pull/21554 is only checking if a real receiver has ever been connected.  It will never return true if only RC overrides have been used.

PR #21925 (2022-10-09): Copter: RC override arming check fix.
Fixes #21921

Not tested.
closed 2022-10-09 by IamPete1