ArduPilot/ardupilot #21932: Skyviper Journey GPS broken

khimaros, 2022-10-10:
## Bug report

**Issue details**

from https://discuss.ardupilot.org/t/gps-broken-on-copter-4-1/78708/3

> Copter 4.1 and master on the SkyViper Journey cannot establish connection to the ublox GPS, status page reports Fixtype 0.
4.0.8 and prior have no such problems.

a fix is apparently available here: https://git.sr.ht/~grone/ardupilot/commit/07380126327422ae8e29e54f81f7b934b12ecdb5

looking at the ardupilot code base, this patch doesn't seem to have been applied.

**Version**
4.1 and master

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad coptor

**Hardware type**
Skyviper Journey

rmackay9, 2022-10-12:
Thanks for the report.  Two other developers have also confirmed that the SkyViper build seems unable to get GPS lock and the LEDs do not seem to work correctly.  I've added this to the Copter-4.3 issues list so that it is less likely to be forgotten.

rmackay9, 2022-10-18:
Hi @khimaros,

We discussed this at the dev call and we think the cause of the issue could be a GPS baud rate change to support higher baud rates by default.  Could you try setting GPS_DRV_OPTIONS to 4 ("Use baudrate 115200")?

rmackay9, 2022-11-07:
@khimaros,

Could you try the fix suggested by @lucasdemarchi?  Just set these two parmeters

- SERVO5_FUNCTION = -1
- SERVO6_FUNCTION = -1

khimaros, 2022-11-08:
@rmackay9 thank you for tackling this!

i incorrectly received a SkyViper Journey SE which is incompatible with ardupilot (as discussed [here](https://discuss.ardupilot.org/t/is-the-sky-viper-journey-se-copter-transmitter-pair-user-upgradeable-to-ardupilot/79275/40)), so i sent it back.

some people in the original `discuss` thread (such as grone@) might be able to verify.

rmackay9, 2022-11-08:
Thanks @khimaros,

I've posted a comment over in [this discussion](https://discuss.ardupilot.org/t/is-the-sky-viper-journey-se-copter-transmitter-pair-user-upgradeable-to-ardupilot/79275/48).  Please tell me if I got the wrong one.

khimaros, 2022-11-08:
@rmackay9 this is the one about GPS issues: https://discuss.ardupilot.org/t/gps-broken-on-copter-4-1/78708/