ArduPilot/ardupilot #21935: Ardupilot 4.3 is incompatible with Ubuntu 22.04.1 LTS

hannes-m, 2022-10-11:
## Bug report

**not compiling at all**

I tried to compile Ardupilot 4.3 on Ubuntu 22.04.1 LTS and run in several problems:
e.g. on the new Ubuntu "Python" is not existing - it is called "Python3" pointing to Python 3.10
The file "make" is in a different directory but not in /usr/bin.
The include-files in the directory "missing" of the cloned repository were not found.
Finally, after one unnecessary long night I got stuck with an error message about "distutils.sysconfig" and gave up.

**Version**
Ardupilot 4.3

**Platform**
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_plane

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2022-10-11:
See setup documentation on the wiki, https://ardupilot.org/dev/docs/building-setup-linux.html#building-setup-linux

closed 2022-10-11 by IamPete1

hannes-m, 2022-10-11:
Hi Peter, Thx a lot...I did not run the prereqs-ubuntu...(I jumped too fast in the documentation)...Successful compilation in 3m54 on a HP Z800.Ì never thought to compile it by myself but I am not very lucky with the airspeedsensor based on differential pressure...I developed  my own one based on a thermal flow sensor which is much more accurate at stall velocity. I already linearized it in a microcontroller and want to integrate it in Ardupilot.Hannes -- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.