ArduPilot/ardupilot #21946: Scripting: crash after scripting restart

IamPete1, 2022-10-12:
It is possible to leave scripting in a state where it will crash on restart. There must be some state somewhere the is not getting reset correctly. This is on 4.2 prior to https://github.com/ArduPilot/ardupilot/pull/20549 which fixed a memory leak and on master. 

This script will cause the issue:
```
logger.write('A', 'A', 'b', 1)
```
It seems that any error in the logger write function once the buffer is created will leave in this sate.

Some stuff that has been tested:

- The running script does not have to crash, we can pcall the logging function and the script carry's on as normal.
- Any script will cause the crash once the first script has been loaded. For example boot with the above script hit the error, then remove the script and replace it with another, `simple_loop.lua` for example. Restarting scripting will still cause a crash. 
- This only happens on real hardware, no luck in SITL. (only tested CubeOrange)
- This issue does not happen if built with `--debug`
- Crash dump results in:
```
warning: Overlapping regions in memory map: ignoring
0x080da144 in lua_scripts::load_all_scripts_in_dir(lua_State*, char const*) [clone .part.0] ()
#0  0x080da144 in lua_scripts::load_all_scripts_in_dir(lua_State*, char const*) [clone .part.0] ()
No symbol table info available.
#1  0x30006c8c in rover ()
No symbol table info available.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```
- watch dog: `WDOG, 129102731, -3, 0, 0, 0, 0, 0, 0, 217, 3, 805478496, 2, 4196355, 1431655765, Scri` gives:
```
    T            Scheduler Task:           -3: Waiting for sample
   IE       Internal Error Mask:            0: ?????
  IEC      Internal Error Count:            0: 0
  IEL       Internal Error Line:            0: 0
   MM           MAVLink Message:            0: [None]
   MC           MAVLink Command:            0: [None]
   SL            Semaphore Line:            0: Not waiting on semaphore
   FL                Fault Line:          217: ?????
   FT                Fault Type:            3: HardFault
   FA             Fault Address:   0x3002a060: ?????
  FTP     Fault Thread Priority:            2: ?????
FICSR        Fault ICS Register:    0x4196355: [Below]
         VECTACTIVE:   3  (Hard fault)
          RESERVED1:   0 
           RETOBASE:   1  (no (or no more) active exceptions)
        VECTPENDING:   0  (Thread mode)
          RESERVED2:   0 
         ISRPENDING:   1  (Interrupt pending)
          RESERVED3:   0 
          PENDSTCLR:   0  (WO clears SysTick exception)
          PENDSTSET:   0  (SysTick not pending)
          PENDSVCLR:   0  (WO clears pendsv exception)
          PENDSVSET:   0  (SysTick not pending)
          RESERVED4:   0 
         NMIPENDSET:   0  (NMI not pending)
          FLR Fault Long Return Address:   0x55555555: ?????
   TN               Thread name:         Scri: ?????
```
That seems to point here: https://github.com/ArduPilot/ardupilot/blob/c801f12a780fd5c75c6a35655ae79d951bb60b8a/libraries/AP_Scripting/lua_scripts.cpp#L217

Spotted while looking for a fix: https://github.com/ArduPilot/ardupilot/pull/21944 https://github.com/ArduPilot/ardupilot/pull/21945

Thanks to @yuri-rage for the original report and help finding a reproducible test case.

tridge, 2022-10-13:
Have you tested valgrind on SITL, the memory checker on chibios and undefined behaviour checker on chibios?

tridge, 2022-10-13:
Also worth checking with gcc 11.3 with debug

andyp1per, 2022-10-13:
Have you tried running with the memory instrumentation on ?

IamPete1, 2022-10-13:
Fixed the valgrind issue, but not the crash.

IamPete1, 2022-10-24:
Should be fixed by https://github.com/ArduPilot/ardupilot/pull/21944