ArduPilot/ardupilot #21950: NMEA GPS granularity

Hwurzburg, 2022-10-13:
Observed while testing Rover 4.2.1 and 4.3, the Blicube NMEA GPS lat/lon is very granular....dont know if its an issue in the Moving Baseline GPS with YAW device itself or in AP....if the GPS is reporting incorrectly, we need to inform BliCube

log: https://www.dropbox.com/s/8w7vkevdasxl4qu/4.2.1-NMEAabs-then-NMEArelative.bin?dl=0

rmackay9, 2022-10-13:
Here's a picture from MP's logs showing the granularity of the GPS (in blue) vs the EKF's nice smooth line (in red)
[image]

WickedShell, 2022-10-14:
I see a lot of what looks like single precision math in the location parsing in the NMEA driver (look at `AP_GPS_NMEA::_parse_degrees()`), I don't have the time to work through the induced accuracy lost, but I could believe it easily creates results as seen above. (NMEA is among the worst GPS drivers we support). It could also be a function of the GPS unit's reporting accuracy as initially theorized.

Hwurzburg, 2022-10-15:
[image]
Tridge's fix worked....closing

closed 2022-10-15 by Hwurzburg