ArduPilot/ardupilot #21984: TECS target altitude changes on its own

Hwurzburg, 2022-10-16:
after an RTL AUTOLAND the CRUISE mode target altitude changes without elev stick input and resists the stick demands on occasion and the initial target altitude when changing into loiter or cruise is sometimes much lower than the entry altitude at mode change

log: https://www.dropbox.com/s/eq5pm2fuvca536y/TECS-hdem-failure.bin?dl=0

Master plus a small gcs message txt change to report airspeed cal offset

shellixyz, 2022-10-18:
I have also seen this issue. The target altitude is offset from the current altitude by a fixed offset every time when switching from a manual throttle mode to cruise. Then after a power cycle it is gone or at least imperceptible

Hwurzburg, 2022-10-18:
@shellixyz if you could rebase your custom code on master, it now has new logging to capture information on target altitude inputs into TECS to help Tridge figure out these issues...a log of this behaviour with those new messages will help him...I cant refly until later this week, myself

shellixyz, 2022-10-18:
@Hwurzburg Yes I saw it I will rebase my code to include the new logging info :+1:

shellixyz, 2022-10-18:
I just flew again this time it was going up every time I switched to cruise

[image]

[log_98_2022-10-18-19-11-18.bin.zip](https://github.com/ArduPilot/ardupilot/files/9813806/log_98_2022-10-18-19-11-18.bin.zip)

Hwurzburg, 2022-10-22:
fixed by #22306

PR #22036 (2022-10-23): AP_NavEKF3: fixed getLLH alt for local origin height
this fixes a bug introduced in #21834

this fix in #21834 was correct for getPosD, but should not have been applied to getLLH

this caused cruise mode in plane to descend/ascend by the difference between the public and local origins on mode entry

fixes #21984
commit 92f25150 (2022-10-23): AP_NavEKF3: fixed getLLH alt for local origin height

this fixes a bug introduced in #21834

this fix in #21834 was correct for getPosD, but should not have been
applied to getLLH

this caused cruise mode in plane to descend/ascend by the difference
between the public and local origins on mode entry

fixes #21984
commit 8e8ceb25 (2022-10-23): AP_NavEKF3: fixed getLLH alt for local origin height

this fixes a bug introduced in #21834

this fix in #21834 was correct for getPosD, but should not have been
applied to getLLH

this caused cruise mode in plane to descend/ascend by the difference
between the public and local origins on mode entry

fixes #21984
closed 2022-10-22 by Hwurzburg