ArduPilot/ardupilot #21989: Gimbal not working / no Types after update from 4.2.3 to 4.3.0

RealTadango, 2022-10-17:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

I updated from Plane 4.2.3 to 4.3.0 and after the update the servo gimbal (for FPV pan / tilt) is not working anymore. In mission planner i cannot select a Gimbal Type (says none and greyed out). Servo assignment is still there.

**Version**

4.3.0 (Was working in 4.2.3)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Flying Wing

**Hardware type**
Mateksys 743 V3

**Logs**
Not relevent since it is a setup issue.

rmackay9, 2022-10-17:
@RealTadango,

The wiki documentation and MP setup screen haven't been updated yet to match the new parameter names.

I'm afraid you'll need to manually set the MNT1_xxx parameters.

RealTadango, 2022-10-18:
I cannot find a lot of parameters (probably my fault) but also there is a bug in the panning code. When moving max left it suddenly outputs max right!. I will go back to 4.2.3 for now.

rmackay9, 2022-12-13:
I think I'm going to close this issue because we're a bit stuck without logs.  I've done some significant updating of [the gimbal wiki pages](https://ardupilot.org/copter/docs/common-cameras-and-gimbals.html) which I hope will help the next user who tries to setup a gimbal.

If we get more information on this particular issue then I'm happy to re-open and re-investigate.

closed 2022-12-13 by rmackay9

RealTadango, 2022-12-13:
I see lots of gimbal fixes in the comming release wo i will test again when 4.3.2 is released.

rmackay9, 2022-12-19:
@RealTadango,

We've found and [fixed the 2nd issue you've reported re panning](https://github.com/ArduPilot/ardupilot/pull/22442).  This will be released in 4.3.3 most likely.

RealTadango, 2022-12-19:
Thanks, i will update and check when the plane has been overhauled. Good to hear it has been found!